CloverHackyColor/CloverBootloader #179: TgtBridge issues with OEM SSDTs

stevezhengshiqi, 2020-06-02:
Hi guys, thank you all for the great contribution to this bootloader!

I got a problem with TgtBridge issue with OEM SSDTs. To be more specific, if I rename a method and use TgtBridge to point to a device not in _SB root tree(for example, _SB.PCI0.HPET), all my OEM SSDTs' methods will be also renamed unexpectedly.

For example, I want to rename HPET._STA to XSTA and set TgtBridge to HPET. However, apart from HPET._STA, all _STA methods in OEM SSDTs are unexpectedly renamed to XSTA. Here's my trouble shooting and related codes in config.plist
```
<dict>
  <key>Comment</key>
  <string>change HPET._STA to XSTA</string>
  <key>Find</key>
  <data>X1NUQQ==</data>
  <key>Replace</key>
  <data>WFNUQQ==</data>
  <key>TgtBridge</key>
  <data>SFBFVA==</data>
</dict>
```
[Trouble Shooting.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4713988/Trouble.Shooting.zip)

For another example, when I want to rename _DSM to XDSM only in my TPD0 device, and I set TgtBridge to TPD0; all the _DSM methods in OEM SSDTs will also be renamed to XDSM, which against my ture purpose. Here's an example code in config.plist that would cause the problem I mentioned.
```
<dict>
  <key>Comment</key>
  <string>change TPD0._DSM to XDSM</string>
  <key>Disabled</key>
  <false/>
  <key>Find</key>
  <data>X0RTTQ==</data>
  <key>Replace</key>
  <data>WERTTQ==</data>
  <key>TgtBridge</key>
  <data>VFBEMA==</data>
</dict>
```

This issue has existed for a long time since r4XXX. Please have a check and thank you all!

stevezhengshiqi, 2020-06-03:
It should be related with the `FixRenameByBridge2`  function in https://github.com/CloverHackyColor/CloverBootloader/blob/c0fba39133cbb95eab2537cbfea849120b200517/rEFIt_UEFI/Platform/FixBiosDsdt.cpp#L1793, but this issue makes no-sense to me as it works perfect in DSDT and only breaks OEM SSDTs. 
cc @goodwin

SergeySlice, 2020-06-05:
~~~
	<key>ACPI</key>
	<dict>
		<key>RenameDevices</key>
		<dict>
			<key>_SB.PCI0.RP03.PXSX</key>
			<string>BRG3</string>
			<key>_SB.PCI0.RP05.PXSX</key>
			<string>TBTG</string>
		</dict>

~~~

stevezhengshiqi, 2020-06-05:
@SergeySlice Thanks for providing an alternative way. But it seems only work for devices? BTW, to avoid using TgtBridge, I can put more characters in Find field to make a more accurate match. Still hope to fix TgtBridge function since I see lots of people use that in their config, which could cause potential problems since their SSDTs would be unintentionally edited.

closed 2020-07-03 by stevezhengshiqi