CloverHackyColor/CloverBootloader #180: Suggestion: Option for adding NVRAM variables

CMMChris, 2020-06-04:
Just a suggestion. It would be nice to have a feature for adding variables into NVRAM like with OpenCore.

This would come very handy for example to enable TRIM on third party SSDs.
Since Catalina the TRIM switch is stored inside the NVRAM and thus TRIM is disabled after every NVRAM clean-up.

Having an option in Clover to automatically add variables to NVRAM could be a handy fix for that annoying issue without the need of patching a Kext.

SergeySlice, 2020-06-16:
This is particular case that can be solved by simple addition to Clover or just by manually adding this variable from Shell. But I see no significant need to make such common feature in Clover: AddVariable/EditVariable/DeleteVariable/UUID/Name/Length/Value/Attributes.

CMMChris, 2020-06-16:
I think it would be useful to be able to add default variables to NVRAM so they are available immediately after NVRAM has been cleared. Anyhow, if you are willing to at least add a feature that allows injecting the TRIM NVRAM variable for Catalina installs that would be a nice addition. It's just so annoying having to run trimforce again after every NVRAM reset. Sometimes I don't even notice NVRAM has been cleared which sometimes happens during macOS updates.
[image]

SergeySlice, 2020-09-19:
Don't reset nvram and set variables from OS or from shell. It is not good to set variables from Clover config.

closed 2020-09-19 by SergeySlice

LeeBinder, 2021-12-02:
I 100% second this request 👍 

Also see [@insanelymac](https://www.insanelymac.com/forum/topic/344454-clover-uefi-secure-boot-and-nvram-emulation/?tab=comments#comment-2772772)