CloverHackyColor/CloverBootloader #182: GUI entry custom icon image overruled by macOS .VolumeIcon.icns

LeeBinder, 2020-06-07:
From my EFI partition which is mainly meant for Clover I also launch a Linux disk image via a custom entry in Clover's GUI section. Clover only chainloads grubx64.efi so I have copied that file (and a few more required ones) to /EFI/boot because a) this does not influence Clover, and b) I do not want a separate partition just for a few Grub2 files.

(just FYI: the Linux image itself sits on my Windows NTFS partition which has its dedicated icon in macOS, a Windows icon via .VolumeIcon.icns)

I enabled custom volume icons in Clover GUI, and assigned Bliss OS' icon 'Bliss-OS_radiant-128x128.icns' in GUI/ Custom Entry/Custom image:
[image]

That used to get displayed correctly in Clover GUI:
[image]

But not anymore after adding a icon to the EFI partition in macOS:
[image]

Now the Bliss OS entry is also displayed with the EFI partition's .VolumeIcon in Clover GUI:
[image]

Choosing the same 128 x 128 icon as a .png instead of the .icns format
[image]

did not make a difference. But after removing the .VolumeIcon for the EFI partition in macOS, on reboot the Bliss OS icon was again displayed correctly in Clover GUI even as a .png. So the icon format does not matter as long as it's correctly 128 x 128.

Even reverting back to Clover_r5108 (latest with the old GUI theming engine) did not make a difference.

I filed this as a bug because I am positive that the priority order should be different - that a custom entry's icon image should NOT be influenced AT ALL by the .VolumeIcon of the volume the entry's bootloader is located at, but that the dedicated icon image should ALWAYS be dominant. It's like, when we assign a custom icon to our macOS Volume, we don't want ALL ICONS on our Desktop to be that same icon, so we .. ;) But right now that's what Clover is doing, which is both, counter intuitive and reverse logical.

In case you feel you want to reproduce this, I attached a zipped up .dmg with all related files.

Thanks for your continuous effort on Clover, and making it more professional with each release!

[EFI (GUI extract).dmg.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4742924/EFI.GUI.extract.dmg.zip)

LeeBinder, 2020-06-09:
Did any of you devs read this? Would already be great if you simply confirm for now. Thank you 👍

vectorsigma72, 2020-06-15:
Hi, I'm not used to using these settings, anyway the image is 128x128 whatever you trying to do:
`Entry->Image.Image.LoadIcns(Volume->RootDir, L"\\.VolumeIcon.icns", 128);`
Maybe you have set the desired image as data:
`
		<key>ImageData</key>
		<data></data>`

LeeBinder, 2020-06-15:
Hi @vectorsigma72 . I converted _Bliss-OS_radiant-128x128.png_ into **base64** via https://www.base64-image.de/ , then in config.plist replaced

```
<key>Image</key>
<string>Bliss-OS_radiant-128x128.png</string>
```

with

```
<key>ImageData</key>
<data>data:image/png;base64,iVBORw0KG[..shortened for readability..]</data>
```

but still the .VolumeIcon.icns of the EFI volume automatically overwrites and is displayed.

Is it correct to insert the string as Base64 or does it have to be something else like e.g. hex? And if Base64 is correct, did I insert the string correctly?

SergeySlice, 2020-06-16:
Why you didn't delete .VolumeIcon.icns if you want Custom image?

LeeBinder, 2020-06-16:
Hi @SergeySlice , thank you for looking into this. The reason why I don't delete .VolumeIcon.icns is because I have several EFI partitions on several hard disks all with dedicated icons for quick recogniton, and all together many partitions also all with dedicated icons for the same reason, so I really would like to keep that stringent.

The logic seems simple and straight forward. Would it be a lot of hassle to either prioritize custom icon images in the code over .VolumeIcon.icns, or to add a query into the existing GUI entry icon display function, some algorithm like:

```
if custom image =! '' then display custom image
   else
      try display .VolumeIcon.icns
      yrt
fi
```
?

SergeySlice, 2020-06-16:
No matter of config.plist VolumeIcon should work so it must be in priority.

LeeBinder, 2020-06-17:
hm well, with that reasoning everything would be hardcoded into the Clover binary and nothing configurable via config.plist .. ;) Naturally and in general, (almost) any setting could be modified from stock Clover binary default settings from within the config.plist.

This notion of "priority" is where the misinterpretation happens which leads to this bug because .VolumeIcon should ONLY display the icon of the actual physical volume as long as no custom icon image is set in a GUI custom entry, *regardless* of the .VolumeIcon. That does not mean at all that Clover would need to abandon or disregard .VolumeIcon - it would just mean an additional query *only* for GUI entries, that's all :)

SergeySlice, 2020-06-18:
Creating the VolumeIcon is a right native method for macOS. Copy image and Paste to disk.
Why we should prefer customising through config.plist? The result is the same.

LeeBinder, 2020-06-18:
sorry but no, it's apparently not the same because you (devs) have coded the custom GUI entry part of Clover in way that allows a custom path to a boot loader file (in this case grub efi), which is a really awesome feature!!! That again reads grub.cgf (also on /EFI) which again can point to not just a partition but (as in this case) to *any* boot image file on *any* partition. Similar to virtualization where you can have xyz VM disk files on one volume, and each VM can have an icon assigned for better recognition. With your current reasoning, **ALL** VM aliases would have the same icon - in this case the one of the EFI partition.

I'm sure you're starting to see now why the current icon image priority for custom GUI entries does not make sense as soon as the entries chainload disk images.

And really, the solution is soooo easy: as soon as a custom (icon) image is set in a custom GUI entry, that overwrites any existing .VolumeIcon **_only for that GUI entry_**. Simple as that :) 🥇

LeeBinder, 2020-07-17:
Hi Slice. Just wondering - did you put any effort into this issue for the 5120 release?

SergeySlice, 2020-07-26:
I have more priority tasks.

LeeBinder, 2020-07-26:
OK. Can you please point me to the two spots in the source code here on github where GUI icons are set in the case of existing
- .VolumeIcon.icns
and
- custom image in a GUI custom entry?

That would be great (if possible even direct links to the two instances).

Thank you Sergey.

SergeySlice, 2020-10-16:
Again. .VolumeIcon.icns will be in priority else delete it.

closed 2020-10-16 by SergeySlice