CloverHackyColor/CloverBootloader #183: Give Instructions

steven-gg, 2020-06-11:
Do please give detailed yet simple instructions on installation and usage. There's little information about the project on the web.
It's hard to find detailed instructions of how to install it on bare metal. I am particularly interestested in using clover with legacy bios systems.

SergeySlice, 2020-06-16:
See old WIKI until there will be new one.
https://sourceforge.net/p/cloverefiboot/wiki/Home/

POTSandSTOP, 2020-08-31:
I'm in a similar situation to @steven-gg -- how do I find out which asset to use / what each one does?

Clover-5121-X64.iso.7z

CloverV2-5121.zip
CLOVERX64.efi.zip

Clover.app_v1.24.pkg
Clover_r5121.pkg

I'M VOLUNTEERING: I'd be glad to take input here and update https://sourceforge.net/p/cloverefiboot/wiki/FAQ/, if you want to temporarily grant me permissions to edit the Wiki over there. I'm one of the project admins for https://sourceforge.net/p/ncid/wiki/Home/ and have been been active in maintaining the Wiki and documentation for it as user [Todd Andrews](https://sourceforge.net/u/tandrews/).

jief666, 2021-04-20:
@POTSandSTOP : volunteering ? We badly need help, for sure, especially in the documentation field.
Do you think the old wiki should be updated, or a new one started ?

joecub31, 2021-05-28:
I very much agree, Linux and macOS hacking information is very vague and not much instruction is given yet people complain when you ask questions about things you're new to. I understand most of y'all have been working on Linux,macOS but some of us that are new to the field and do not know or study something else other than technology would like to figure it out and get some help on on the way￼.

jief666, 2021-05-29:
Not sure what you agreed to. The lack of documentation ?
I know documentation is mainly googling forums. Especially if you show that you tried to search by yourself, at least a little, I’m sure you’ll get help on the main Clover forum.
Just avoid saying « please help » and nothing more. We are not psychics and we can’t guess.

VincentBentley, 2021-06-10:
I will start by saying thanks for all the hard effort that has gone into Clover. I would really love to be able to use it so that I can get an NVME drive for L2ARC on my ZFS servers. But documentation, and not having a modern monitor has got me beat (It's a shame HiRes laptops can't be used as emergency monitors).

So the documentation says that the default screen resolution is 1024x768 which would be fantastic as that is the maximum resolution of my LCD monitor. However, when I boot Clover from a freshly burned CD, my monitor shuts down because its obvious that 1024x768 is no longer the default. No problem, I've modified CD images before...

After loop mounting the ISO image I can see that the default resolution in the 5136 release is now 1280x1024. I have changed it back to 1024x768 but I cannot write out a new mkisofs image because I don't know what bootloader files need to be inside it or what any of the options are. 

So I try to make a bootable USB instead assuming that the 5136 ISO is hybridized for writing to a USB flash drive. I 'dd' one on Ubuntu. Hmm, it doesn't boot. Make one on Windows using Balena Etcher, that doesn't boot either. Then I check with fdisk -l and find the image is not hybridized for USB. I don't have a Mac any more so cannot try that. I'm at a dead end.

I would like to modify the 5136 ISO to include the modified config file that I have created which just has 1024x768 for the screen resolution. I need to know how I can create a new ISO image file that includes my change. I can do this on intel/amd Linux, FreeBSD, OpenIndiana or Windows10, but I don't have access to a Mac.

Alternatively, if someone can tell me how I can create a bootable USB flash drive with the 5136 release that would be equally wonderful assuming I can still modify the config file to what I need on the USB drive.

Thanks

VincentBentley, 2021-06-11:
> So the documentation says that the default screen resolution is 1024x768 which would be fantastic as that is the maximum resolution of my LCD monitor. However, when I boot Clover from a freshly burned CD, my monitor shuts down because its obvious that 1024x768 is no longer the default.

I found my own solution using an old technique that I used many decades ago. I used a hex editor to patch the ISO directly. I replaced '1280x1024</string>' with '1024x768</string> '. As '768' is one digit shorter than '1024', I shuffled forward the closing string tag and put a space character following its closing '>'. It works, and I know I can just as easily patch future releases with the same technique.

closed 2021-07-06 by jief666