CloverHackyColor/CloverBootloader #186: Clover will not update; "Installation type" tab not available

sandymc, 2020-06-15:
1. Attempting to update from 5070 to 5119 using the package from this site

2. Update runs without error, but nothing installed to ESP - files are not updated, and version number on boot remains 5070

3. "Installation type" tab never appears; the installer skips right past

4. Installation refers to doing a "standard install"

It seems that probably the installer is incorrectly doing a non-UEFI install. But there appears to be no (obvious) way to change this.

vectorsigma72, 2020-06-15:
Use [Clover.app](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5119/Clover.app_v1.23.pkg).

sandymc, 2020-06-15:
> Use [Clover.app](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5119/Clover.app_v1.23.pkg).

That is what I'm using.

vectorsigma72, 2020-06-15:
> > Use [Clover.app](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5119/Clover.app_v1.23.pkg).
> 
> That is what I'm using.

Sorry but Clover.app has no "Installation type tab”:
[image]
As you can see there are no tabs.

sandymc, 2020-06-15:
That's not what I get.
[image]

vectorsigma72, 2020-06-15:
Did you realize it is an application to install Clover and not a pkg to install Clover? 
After the installation you can find an /Application/Clover.app. Look at the top bar then:
[image]
then again push on "Install Clover r5xxx” 
[image]

P.S. the first time it downloads Clover from Github/releases and then you're ready, so give it the time to download it.

sandymc, 2020-06-15:
Thank you, that resolved the issue. However, might I suggest that you revert to an installer that directly installs the package. This two-step process is not particularly user friendly.

closed 2020-06-15 by sandymc

vectorsigma72, 2020-06-15:
> Thank you, that resolved the issue. However, might I suggest that you revert to an installer that directly installs the package. This two-step process is not particularly user friendly.

Sorry but both the package and the app are available, so just download what you prefear. Nothing to revert.