CloverHackyColor/CloverBootloader #188: Error: Not Found while LegacyBoot

serdeliuk, 2020-06-19:
Hello,

I am trying to boot ms-dos from a usb stick and it seems impossible, clover see the stick but is unable too boot generating following error and return to boot menu options after i hit any keys.

Error: Not Found while LegacyBoot
was error, press any key
* Hit any key to continue *

Does work this legacy boot? My notebook's UEFI bios have almost the same behavior see the stick but is not able to boot from, the stick boot on other (old) notebook so there is nothing worng with the stick.

Is there anything i can do or test?

Best regards,

Marc

SergeySlice, 2020-06-29:
ms-dos??? Impossible.

serdeliuk, 2020-06-29:
:))) well, i need to boot ms dos, unfortunately i was unable to boot, my bios even has legacy boot does not boot, clover even ha one too give that "cryptic / unusual "error and even grub refuze to chainload ms-dos..... 

However, the error message is not very well suited for the error :) at least that can be changed.

closed 2020-09-19 by SergeySlice