CloverHackyColor/CloverBootloader #190: Cannot boot for macOS 10.16 installer

sopanus2014, 2020-06-23:
Cannot boot for macOS 10.16 installer

cubem10, 2020-06-23:
Me too.. It stucks on Error 0xE, 0 @ LocHB ~~~ and showing date and +++++... after Err(0xE) @ OPEN (usr\\standalone\\DS.dmg_root_hash)
This EFI worked on Catalina 10.15.5

gilberthdez, 2020-06-23:
macOS 11* :D

DMNerd, 2020-06-23:
This might have something to do with the new kernel reworks. Hope this will be able to get resolved :)

ManasMalla, 2020-06-23:
yah it stops at LOG:EXITBS:START ! Waiting for the fixes ASAP

eskimo, 2020-06-24:
> macOS 11* :D

[10.16?](https://i.arxius.io/7ac019eb.png)

mifjpn, 2020-06-24:
mee to!

JoyZgq, 2020-06-24:
wait for it

SalvaMontepaone, 2020-06-28:
nothing new?

SergeySlice, 2020-06-28:
You seems want to be the first who discover America.
Yes, we already known that the Clover can't boot BigSur installer. What is next step? Open the issue???

SalvaMontepaone, 2020-06-28:
> You seems want to be the first who discover America.
> Yes, we already known that the Clover can't boot BigSur installer. What is next step? Open the issue???

Sorry I just asked if there was nothing new  :(

ITzTravelInTime, 2020-06-29:
I think that we should leave slice to do his job as clover developer, i just hope that he can figure the new os out before the end of the summer and release an updated clover.

Too bad i know so little about macos architecture and bootloaders, because i know quite well the c languages for low level usage, i have done some standalone ms-dos programs in this way.

ITzTravelInTime, 2020-06-29:
About the clover issues the new os changed quite a lot about the kernelcache architetcure and how the boot process of the os works, it seems like a lot of stuff about clover and it's uefi drivers needs some very huge rework, so as one of the few developers still working on this project slice has a lot of difficoult stuff to deal with so be gentile with him and don't give him pressure.

I whould also thank slice for all the work he is doing for us.

closed 2020-06-29 by sopanus2014