CloverHackyColor/CloverBootloader #200: Catalina 10.15.5 won't boot after upgrading to Clover 5120

brunoaduarte, 2020-07-17:
I've just updated to Clover 5120 and now it won't boot anymore.

Here are the files that were changed on the upgrade.

[image]

norton287, 2020-07-17:
After updating mine just hangs on the row of +++++++++++++++++ signs 3 out of 4 times before finally booting.  I'm not sure if I need to set up those quirks in the loader menu or not to resolve the issue.

brunoaduarte, 2020-07-17:
Rolledback to 5119, upgrade Catalina to 10.15.6 and tried 5120 again, error persists.

[image]

debug.log
```
0:100  0:100  MemLog inited, TSC freq: 3192451370
0:100  0:000  CPU was calibrated with RTC
0:100  0:000  
0:100  0:000  Now is 17.07.2020,  17:34:17 (GMT)
0:100  0:000  Starting Clover revision: 5120 (master, commit df946b75f) on American Megatrends EFI
0:100  0:000  Build with: [Args: -D NO_GRUB_DRIVERS_EMBEDDED -t GCC53 | -D NO_GRUB_DRIVERS_EMBEDDED --conf=/Users/sergey/src/Clover/Conf -D USE_LOW_EBDA -a X64 -b RELEASE -t GCC53 -n 5 | OS: 10.14.6]
0:100  0:000  SelfDevicePath=PciRoot(0x0)\Pci(0x17,0x0)\Sata(0x2,0xFFFF,0x0)\HD(1,GPT,D5C47FCF-C297-4F31-B34F-6096EA0D03AE,0x28,0x64000) @880BEF98
0:100  0:000  SelfDirPath = \EFI\BOOT
0:101  0:001  Clover : Image base = 0x7DAED000
0:101  0:000  SimpleTextEx Status=Success
0:101  0:000  === [ Get Smbios ] ==============================
0:101  0:000  Type 16 Index = 0
0:101  0:000  Total Memory Slots Count = 4
0:101  0:000  Type 17 Index = 0
0:101  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:101  0:000  SmbiosTable.Type17->Size = 8192MB
0:101  0:000  SmbiosTable.Type17->Bank/Device = BANK 0 ChannelA-DIMM1
0:101  0:000  SmbiosTable.Type17->Vendor = G-Skill
0:101  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:101  0:000  SmbiosTable.Type17->PartNumber = F4-2400C15-8GVR     
0:101  0:000  Type 17 Index = 1
0:101  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:101  0:000  SmbiosTable.Type17->Size = 8192MB
0:101  0:000  SmbiosTable.Type17->Bank/Device = BANK 1 ChannelA-DIMM2
0:101  0:000  SmbiosTable.Type17->Vendor = G-Skill
0:101  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:101  0:000  SmbiosTable.Type17->PartNumber = F4-2400C15-8GVR     
0:101  0:000  Type 17 Index = 2
0:101  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:101  0:000  SmbiosTable.Type17->Size = 8192MB
0:101  0:000  SmbiosTable.Type17->Bank/Device = BANK 2 ChannelB-DIMM1
0:101  0:000  SmbiosTable.Type17->Vendor = G-Skill
0:101  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:101  0:000  SmbiosTable.Type17->PartNumber = F4-2400C15-8GVR     
0:101  0:000  Type 17 Index = 3
0:101  0:000  SmbiosTable.Type17->Speed = 2133MHz
0:101  0:000  SmbiosTable.Type17->Size = 8192MB
0:101  0:000  SmbiosTable.Type17->Bank/Device = BANK 3 ChannelB-DIMM2
0:101  0:000  SmbiosTable.Type17->Vendor = G-Skill
0:101  0:000  SmbiosTable.Type17->SerialNumber = 00000000
0:101  0:000  SmbiosTable.Type17->PartNumber = F4-2400C15-8GVR     
0:101  0:000  Boot status=0
0:101  0:000  Running on: 'System Product Name' with board 'TUF Z390-PLUS GAMING (WI-FI)'
0:101  0:000  === [ GetCPUProperties ] ========================
0:101  0:000  CPU Vendor = 756E6547 Model=906EA
0:101  0:000   The CPU supported SSE4.1
0:101  0:000  BrandString = Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz
0:101  0:000   The CPU supported turbo
0:101  0:000  MSR 0x35               6000C
0:101  0:000   TSC/CCC Information Leaf:
0:101  0:000    numerator     : 266
0:101  0:000    denominator   : 2
0:101  0:000   Calibrated ARTFrequency: 24003393
0:101  0:000   Rounded ARTFrequency: 24000000
0:101  0:000  MSR 0xE2 before patch 1E000000
0:101  0:000  MSR 0xCE              08080838_F1012000
0:101  0:000  corrected FLEX_RATIO = 0
0:101  0:000  MSR 0x1B0             00000000
0:101  0:000  FSBFrequency = 100 MHz, DMI FSBFrequency = 100 MHz, Corrected FSBFrequency = 100 MHz
0:101  0:000  MaxDiv/MinDiv: 32.0/8
0:101  0:000  Turbo: 43/44/45/46
0:101  0:000  Features: 0x77FAFBFFBFEBFBFF
0:101  0:000  Threads: 12
0:101  0:000  Cores: 6
[... 2247 lines cut ...]
32:108  0:019  ->Extra kext: EFI\CLOVER\kexts\Other\AMDRadeonX4000.kext (v.3.0.8)
32:162  0:053  ->Extra kext: EFI\CLOVER\kexts\Other\USBInjectAll.kext (v.0.7.5)
32:477  0:315  ->Extra kext: EFI\CLOVER\kexts\Other\SMCProcessor.kext (v.1.1.4)
32:496  0:019  ->Extra kext: EFI\CLOVER\kexts\Other\GenericUSBXHCI.kext (v.1.2.11)
32:516  0:020  ->Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.4.0)
32:540  0:023  ->Extra kext: EFI\CLOVER\kexts\Other\SMCSuperIO.kext (v.1.1.4)
32:558  0:018  ->Extra kext: EFI\CLOVER\kexts\Other\IntelMausiEthernet.kext (v.2.5.1d1)
32:576  0:017  ->Extra kext: EFI\CLOVER\kexts\Other\IntelBluetoothFirmware.kext (v.1.1.1)
32:692  0:115  ->Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.5.0)
33:488  0:796  ->Extra kext: EFI\CLOVER\kexts\Other\itlwm.kext (v.1)
33:604  0:116  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.4.5)
33:627  0:023  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Off
33:634  0:006  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10
33:641  0:006  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10_normal
33:650  0:009  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15
33:656  0:006  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15_normal
33:663  0:006  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6
33:670  0:006  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6_normal
33:691  0:021  Custom boot screen not used because entry has unset use graphics
33:698  0:006  Closing log
33:705  0:006  start image 'boot.efi'
33:720  0:014  Using load options 'boot.efi dart=0 -v '
33:757  0:037  OSInfo:OSName called

```

LeeBinder, 2020-07-17:
So which Catalina now? The recent update was 10.15.6 not .5

If you have issues with 10.15.5, why don't you update to 10.15.6?

brunoaduarte, 2020-07-17:
> So which Catalina now? The recent update was 10.15.6 not .5
> 
> If you have issues with 10.15.5, why don't you update to 10.15.6?

I've upgraded Catalina to 10.15.6 after my post, but it doesn't matter, the problem persists. The problem is with Clover bootloader upgrade from 5119 to 5120. Not related to Catalina version.

eduardoborges, 2020-07-18:
Same problem here

LeeBinder, 2020-07-18:
@brunoaduarte please then change the topic title accordingly into

**Catalina (10.15.5/.6) won't boot after upgrading to Clover 5120**

because no matter if you edit 10.15.5 and 10.15.6 back and forth in the title is misleading as you verified yourself .. ;) The devs need us to be as precise as possible for trouble shooting -> issue solving. Thank you.

To all incl. @eduardoborges : my Catalina wasn't booting either after the 5120 update (Mojave and High Sierra still were). I was able to narrow it down to:

**`/Volumes/EFI/EFI/BOOT/BOOTX64.efi`**

So please, with Clover 5120 installed, rename the file to e.g. `BOOTX64.efi.5120` and copy BOOTX64.efi from your main macOS volume's

**`/EFI-Backups/r5119/2020-07-xy-xyhxy/EFI/BOOT/BOOTX64.efi`**

into that folder, reboot, and report back if for you also it's that file that affects booting, so Slice (or other devs) can hopefully solve this easier.

Thanks.

jhsvip, 2020-07-18:
I used the latest development version of openruntime and lilu weg virstulsmc
5120 update All works in 10.15.6 fine，but bigsur still does not work well。

When I boot bigsur
A purple stripe will appear on the logo screen（RX580 8G，I guess it hasn't driven）
After that It stucks in a black screen with a mouse pointer

brunoaduarte, 2020-07-18:
This seems to be totally random. I've installed R5120 again and on the first boot it worked perfectly. I changed nothing and just restarted it again, and it freezes after AppleKeyStore message. Restarted again and it freezes again. Restarted for the 4th time and it WORKED!
1) Book OK
2) Freeze
3) Freeze
4) Boot OK
That makes no sense.

LeeBinder, 2020-07-18:
@brunoaduarte yeah, I forgot to mention that I experienced it the exact same way. I just did 4 boots, 3 of which got stuck.

Thanks for changing the issue title. I encourage you to test if for you also 5119 BOOTX64.efi makes your Catalina boot again all the time.

@jhsvip can you please zip up your latest self-compiled versions of openruntime lilu weg and virtualsmc, and attach the zip to your next reply here? That would be very helpful 👍

brunoaduarte, 2020-07-18:
> @brunoaduarte yeah, I forgot to mention that I experienced it the exact same way. I just did 4 boots, 3 of which got stuck.
>
> Thanks for changing the issue title. I encourage you to test if for you [...]

Yes, replacing EFI/EFI/BOOT/BOOTX64.efi with the one from R5119 solves the problem.

And i did some exaustive tests here, it's confirmed, for every 2 bad boots (freeze after AppleKeyStore: operation failed), it will boot 1 time perfectly. Very weird behavior.

LeeBinder, 2020-07-18:
Allright, good work. And congrats you can confirm and got it flowing again. Let's wait now for the kexts + efi drivers from jhsvip.

jhsvip, 2020-07-19:
> Allright, good work. And congrats you can confirm and got it flowing again. Let's wait now for the kexts + efi drivers from jhsvip.
Kexts here.
[Kexts_2.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4943433/Kexts_2.zip)
openruntime.efi You can download from https://github.com/williambj1/OpenCore-Factory

norton287, 2020-07-19:
Still hanging on boot at a row of ++++++++++++++++++ signs after making changes from jhsvip recommendations.  I'm just lost as what is wrong.

LeeBinder, 2020-07-19:
@jhsvip thanks for your kexts - will try soon

@norton287 if you haven't done so already, study the screenshot jhsvip provided for us:

<details>
  <summary>click to expand</summary> 

  [image]
---
</details>

In addition to OpenRuntime.efi, you might also need Bootstrap.efi (included in the OC release zip package) and choose OcQuirks.efi during Clover install:

[image]

then activated the quirks in clover.plist as per screenshot, which you can do with the latest Clover Configurator.

**<ins>ATTENTION</ins>**: if you use [OCQuirks](https://github.com/ReddestDream/OcQuirks), you have to make sure there is neither `AptioMemoryFix` nor any `OsxAptioFix` variant present in drivers/UEFI - remove any such drivers if there is one after Clover 5120 install and before rebooting!

Did that work for you now?

norton287, 2020-07-19:
Did everything and no luck here is what I have:

On July 1

Here is my quirks section of my plist:

click to expand

ATTENTION: if you use OCQuirks, you have to make sure there is neither AptioMemoryFix nor any OsxAptioFix variant present in drivers/UEFI - remove any such drivers if there is one after Clover 5120 install and before rebooting!

LeeBinder, 2020-07-19:
@norton287 your comment got screwed, look.

Apart from that, same thing here: I step-by-step adapted my Clover config accdg. to jhsvip's screenshot in many reboots until it even matched 100% (except IntelMausEthernet.kext which I don't need), even Clover 5120 OpenRuntime.efi replaced with the lates downloaded from OpenCore Factory - same issue: "`AppleKeyStore: operation failed`", then IGPU does not load, and loops. Mojave again still boots just fine.

@jhsvip there is an error in your drivers/UEFI folder: you need to remove SMCHelper.efi (working with FakeSMC.kext) because you use VirtualSMC.kext which replaced FakeSMC.kext. Maybe then even Big Sur might run.. ;)

jhsvip, 2020-07-20:
My config.plist here.
[config.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4946278/config.zip)

LeeBinder, 2020-07-20:
Thanks @jhsvip - might take me until next week-end to look at it. Did you see my hint for you re. SMCHelper.efi?

jhsvip, 2020-07-20:
> Thanks @jhsvip - might take me until next week-end to look at it. Did you see my hint for you re. SMCHelper.efi?
Thanks For your hint!
 Yes,SMCHelper.efi replaced to VirtualSMC.efi, But nothing changed.

LeeBinder, 2020-07-20:
Oh from what I remember VirtualSMC.efi is obsolete, no need for it eiter :)

norton287, 2020-07-20:
Here is what I have as far as my non-working config in 5120:

[image]

[image]

![Screen Shot 2020-07-19 at 2 56 38 PM](https://user-images.githubusercontent.com/42098364/87935089-03ebba80-
ca56-11ea-832d-83344fe03233.png)

I have disabled SMCHelper.efi since and tried again with no luck.

John

LeeBinder, 2020-07-20:
yeah, the quirks sure don't cut it, they're not the "do-jobby". Last chance I think would be to test jhsvip's config.plist, either as-is (probably won't work unless your hardware is very close to his), or to copy sections form his plist over to yours one at a time and test-boot Catalina. I found that to be easiest with PrefEdit.

Needless to say also chances are it's hardware related, and jhsvip simply lucked out with his hack rig.

SergeySlice, 2020-07-20:
Test updated 5120.

jhsvip, 2020-07-20:
[image]
[image]

It works now

changes here
NVRAM：
[image]
Patch：
[image]

GoodLuck！

norton287, 2020-07-20:
was able to install with new 5120, but when I boot into the system with my Saphire Pulse RX580 8GB I go to a black screen when it supposed to switch to the apple logo and never shows the desktop.  Any ideas, I have agdpmod=pikeralpha since Catalina still configured.

John

brunoaduarte, 2020-07-20:
Latest commits fixed the issue here on Catalina 10.15.6. Boots everytime now.

LeeBinder, 2020-07-20:
@SergeySlice wrote:
> Test updated 5120.

yes, thank you Sergey for your efforts - now booting every time, **_BUT_** the randomness just shifted to something new: now every other time VoodooI2C does not load (from EFI/CLOVER/kexts) and the trackpad pointer does not move. With 5119 it would always load.

I already ran `rebuild kextcache -i` and rebooted but that did not solve it.

All of this seems like some newly introduced timing issue? Do you need some log?

LeeBinder, 2020-07-20:
@jhsvip : great you got Big Sur going, and thanks for sharing your NVRAM changes + patch!

@norton287 : I kindly suggest you switch over to the Big Sur issue so this thread stays on topic :)

jhsvip, 2020-07-21:
> was able to install Big Sur with new 5120, but when I boot into the system with my Saphire Pulse RX580 8GB I go to a black screen when it supposed to switch to the apple logo and never shows the desktop. Any ideas, I have agdpmod=pikeralpha since Catalina still configured.
> 
> John

If you enter the desktop after startup, the screen will be blank. Please do the following:
Enter Mac OS 10.15 or enter the installation mode terminal and execute the following command:
sudo nvram booter-fileset-kernel="%00"
Check whether the assignment is successful:
nvram -p |grep booter-fileset-kernel
booter-fileset-kernel %00

If successful, the reboot will go smoothly to the desktop

norton287, 2020-07-21:
@jhsvip Thanks so much that worked perfectly!

[image]

LeeBinder, 2020-07-21:
@jhsvip @norton287 see attachment :) Feel free to spread in Big Sur boot (issues) threads.

[cure Big Sur black screen.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4956943/cure.Big.Sur.black.screen.zip)

closed 2020-07-26 by SergeySlice