CloverHackyColor/CloverBootloader #202: Autoboot stopped working after upgrade from 5104 to 5119

brunoaduarte, 2020-07-17:
After i've upgraded from Clover 5104 to 5119 my autoboot stopped working. My config.plist is correct and volume name has not changed. If i rollback to 5104 it works again.

config.plist

```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>ACPI</key>
	<dict>
		<key>AutoMerge</key>
		<true/>
		<key>DSDT</key>
		<dict>
			<key>#DropOEM_DSM</key>
			<dict>
				<key>ATI</key>
				<true/>
				<key>Firewire</key>
				<true/>
				<key>HDA</key>
				<true/>
				<key>HDMI</key>
				<true/>
				<key>IDE</key>
				<true/>
				<key>IntelGFX</key>
				<true/>
				<key>LAN</key>
				<true/>
				<key>LPC</key>
				<false/>
				<key>NVidia</key>
				<true/>
				<key>SATA</key>
				<true/>
				<key>SmBUS</key>
				<false/>
				<key>USB</key>
				<true/>
				<key>WIFI</key>
				<true/>
			</dict>
			<key>Debug</key>
			<false/>
			<key>Fixes</key>
			<dict>
				<key>AddDTGP</key>
				<false/>
				<key>AddHDMI</key>
				<false/>
				<key>AddIMEI</key>
				<false/>
				<key>AddMCHC</key>
				<false/>
				<key>AddPNLF</key>
				<false/>
				<key>DeleteUnused</key>
				<false/>
				<key>FakeLPC</key>
				<false/>
				<key>FixACST</key>
				<false/>
				<key>FixADP1</key>
				<false/>
				<key>FixAirport</key>
				<false/>
				<key>FixDarwin</key>
				<false/>
[... 914 lines cut ...]
8:889  0:006  ->Extra kext: EFI\CLOVER\kexts\Other\SMCBatteryManager.kext (v.1)
8:896  0:007  ->Extra kext: EFI\CLOVER\kexts\Other\AHCI_Intel_Generic_SATA.kext (v.0.5)
8:897  0:000  ->Extra kext: EFI\CLOVER\kexts\Other\AMD9500Controller.kext (v.3.0.8)
8:956  0:059  ->Extra kext: EFI\CLOVER\kexts\Other\VirtualSMC.kext (v.1.1.4)
8:968  0:011  ->Extra kext: EFI\CLOVER\kexts\Other\AMDRadeonX4000.kext (v.3.0.8)
9:268  0:300  ->Extra kext: EFI\CLOVER\kexts\Other\USBInjectAll.kext (v.0.7.1)
9:317  0:049  ->Extra kext: EFI\CLOVER\kexts\Other\SMCProcessor.kext (v.1.1.4)
9:327  0:010  ->Extra kext: EFI\CLOVER\kexts\Other\GenericUSBXHCI.kext (v.1.2.11)
9:337  0:009  ->Extra kext: EFI\CLOVER\kexts\Other\RealtekRTL8111.kext (v.2.2.2)
9:361  0:023  ->Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.4.0)
9:391  0:030  ->Extra kext: EFI\CLOVER\kexts\Other\SMCSuperIO.kext (v.1.1.4)
9:405  0:013  ->Extra kext: EFI\CLOVER\kexts\Other\IntelMausiEthernet.kext (v.2.5.0)
9:418  0:013  ->Extra kext: EFI\CLOVER\kexts\Other\AtherosE2200Ethernet.kext (v.2.3.0)
9:431  0:013  ->Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.5.0)
9:604  0:173  ->Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.4.5)
9:615  0:010  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Off
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10_normal
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15_normal
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6
9:615  0:000  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.15.6_normal
9:618  0:003  SetScreenResolution: 1920x1080 - already set
9:618  0:000  Custom boot is disabled
```

brunoaduarte, 2020-07-20:
I did some tests here and the autoboot feature broke on R5105. It was fine on R5104.

https://github.com/CloverHackyColor/CloverBootloader/compare/5104...5105

SergeySlice, 2020-07-20:
Autoboot is working for a while at all other users. Search your mistake.
You are using SSDT-PMC so I have to propose you have working native NVRAM. If so you have to exclude EmuVariableUEFI.
And then go to System Preferences, open Startup Disk and choose Catalina. Reboot.

closed 2020-07-20 by SergeySlice

brunoaduarte, 2020-07-20:
I tried the proposed solution of excluding EmuVariableUEFI from drivers and drivers64UEFI folders -> System Preferences, open Startup Disk -> Catalina and Reboot. And it works, but only for the next startup. If i restart normally on the next times thru the Apple icon it will still not autoboot.

Also this is a duplicate of #93 that is also unsolved. It doesn't seem like a "user's mistake" if something broke after a software upgrade where no other variable changed (nothing changed on my system or config files).

As i said, it works perfectly with 5104, but stops working with 5105... and is the same problem @Razor004 had in his ticket.
Also i have the same SSD setup (2 SSDs, one for Windows 10 and other for MacOS).
And also if i remove the Windows SSD autoboot starts working again! Just like @Razor004 stated.
Something related to that clearly changed on the code.

If you want to replicate the problem it's very simple, install a bootable Windows 10 SSD on the same computer along with MacOS SSD and try R5105, you will see it won't autoboot. Then downgrade to 5104 and see it autobooting.

Maybe most users don't have this problem because they don't have another SSD installed in their systems or the problem just won't bother them. The problem is real.