CloverHackyColor/CloverBootloader #203: Новый кловер r5120 после обновления ломает загрузку 10.15.6

feedlike, 2020-07-17:
Почему после обновления кловера он ломает загрузку системы 10.15.6? обновляю путем накатывания через pkg на efi раздел системы

vectorsigma72, 2020-07-17:
Если у вас был OpenRuntime.efi, pkg удалил его по ошибке. Проблема решена, pkg был заменен.

ARGAMX, 2020-07-17:
Проапдейтился с r5119 на r5120, macOS 10.15.4 перестала грузится. Висит яблоко бесконечно, прогресса загрузки нет. Откатился обратно - работает.

feedlike, 2020-07-17:
> Если у вас был OpenRuntime.efi, pkg удалил его по ошибке. Проблема решена, pkg был заменен.

OpenRuntime.efi не было до этого, в чем может быть проблема?

vectorsigma72, 2020-07-18:
> > Если у вас был OpenRuntime.efi, pkg удалил его по ошибке. Проблема решена, pkg был заменен.
> 
> OpenRuntime.efi не было до этого, в чем может быть проблема?

Возможно некоторые изменения в kernelpatcher, которые работают только для нескольких. Это будет исправлено.

LeeBinder, 2020-07-18:
see and try what helped me --> https://github.com/CloverHackyColor/CloverBootloader/issues/200#issuecomment-660452197

SergeySlice, 2020-07-20:
Исправлено

closed 2020-07-20 by SergeySlice

feedlike, 2020-07-20:
> Исправлено

Протестирую на днях. Спасибо @SergeySlice