CloverHackyColor/CloverBootloader #207: Can't see Internal Drives after Successfully booting to Big Sur installer

hohoho123456, 2020-07-21:
I was able to boot to the Big Sur Installer using the latest clover 5120, but once i got into the installer, i can't see my internal drives. i have a 1 TB HD and an Sata ssd and both of them are not present in disk utility and terminal so I can't install Big Sur. I also tried to update my Catalina to Big Sur using the same exact USB, but the update to Big Sur never continues and is just gives me a stop sign. But when i use the same exact USB to boot Catalina it boots Catalina perfectly. This same exact problem also exist when I try to use Opencore to boot to the installer, if that information helps too. Thanks in advance for the help!
[clover 5120 config.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4950679/clover.5120.config.zip)
[image]
[image]
[image]
[pcidevices.txt](https://github.com/CloverHackyColor/CloverBootloader/files/4950686/pcidevices.txt)

qw7575, 2020-07-21:
5120可以装大苏了吗

hohoho123456, 2020-07-21:
> 5120可以装大苏了吗

No. Can't install Big Sur. Big Sur installer is not detecting my internal Sata SSD.

closed 2020-07-23 by hohoho123456