CloverHackyColor/CloverBootloader #208: 再次提交的5120可以装大苏吗，是否正常启动卡特琳娜版本

qw7575, 2020-07-21:
如题
根据上周六取消的5120
现在再次提交的5120可以装大苏吗，
是否正常启动卡特琳娜版本

elton11220, 2020-07-21:
Catalina正常启动，但是Big Sur我没进去

jhsvip, 2020-07-21:
> 如题
> 根据上周六取消的5120
> 现在再次提交的5120可以装大苏吗，
> 是否正常启动卡特琳娜版本

Look at my last config.plist, it can boot BigSur normally.
https://github.com/CloverHackyColor/CloverBootloader/issues/200

elton11220, 2020-07-21:
> > 如题
> > 根据上周六取消的5120
> > 现在再次提交的5120可以装大苏吗，
> > 是否正常启动卡特琳娜版本
> 
> Look at my last config.plist, it can boot BigSur normally.
> #200

How did you apply the features in the 5120 to your original clover configuration?
I replaced clover64.efi, boot64.efi, and added the built in kernel patches to my config.plist.
But it still doesn't work in macOS 10.16(10.15.6 works well)

jhsvip, 2020-07-21:
> How did you apply the features in the 5120 to your original clover configuration?
> I replaced clover64.efi, boot64.efi, and added the built in kernel patches to my config.plist.
> But it still doesn't [...]

Plist Editor Or Clover.app
Look at my post

closed 2020-09-19 by SergeySlice