CloverHackyColor/CloverBootloader #211: How to set booter-fileset-kernel="%00" in Clover config?

jardenliu, 2020-07-23:
Hello,
     Yesterday, I upgrade my clover EFI to 5120.  And set quirks options, and success to upgrade my laptop to Big Sur beta2. Before  upgrade, I have to exec `sudo nvram booter-fileset-kernel="%00"`. Is Clover possible to provide settings for this ?

Thanks!

jhsvip, 2020-07-24:
> Hello,
> Yesterday, I upgrade my clover EFI to 5120. And set quirks options, and success to upgrade my laptop to Big Sur beta2. Before upgrade, I have to exec `sudo nvram booter-fileset-kernel="%00"`. Is Clover possible to provide settings for this ?
> 
> Thanks!

At the moment, I don't find a way to add it to clover.
Using bigsur system requires adding this setting, which should be a defect of clover. OpenCORE can start bigsur without adding this setting.
I hope developers can fix this problem in the next version of clover. Adding NVRAM is too cumbersome.

SergeySlice, 2020-07-26:
Once  you add this variable you need no to repeat this again.
Not need to set this variable at every boot.

jhsvip, 2020-07-26:
> Once you add this variable you need no to repeat this again.
> Not need to set this variable at every boot.

Thanks for the replay！
I want to improve the clover user experience

Make it easier to use, rather than requiring cumbersome configuration

I don't know how opencore can run bigsur without adding that variable, but it did.

closed 2020-09-19 by SergeySlice