CloverHackyColor/CloverBootloader #212: Unable to boot Big Sur Beta 3 prebooter/installer

norton287, 2020-07-23:
It hangs after a bunch of messages about not being able to load kexts and then errors out on loading NVME driver from apple.  Here is a screenshot of my boot screen:

[image]

This is my EFI

[EFI.zip](https://github.com/CloverHackyColor/CloverBootloader/files/4965793/EFI.zip)

Any help would be appreciated

John

ITzTravelInTime, 2020-07-23:
You have a CPU pm isse and virtualsmc failed to load, those (expecially the first one) are problems which can cause a system not to boot, probably you cpu power management isn't native or you have configured your clover poorly

xiaozhangv, 2020-07-23:
me too.any err about can't load kexts

norton287, 2020-07-23:
My config worked fine for installing Beta 2 no problem, only had to add an NVRAM variable to actually boot into Big Sur.

norton287, 2020-07-24:
@jhsvip did you have any luck upgrading to Big Sur Beta 3?

jhsvip, 2020-07-24:
> @jhsvip did you have any luck upgrading to Big Sur Beta 3?

I  have not try to use clover to boot beta3。
Try to use fakesmc to replace virstulSMC。

norton287, 2020-07-24:
Will try that

norton287, 2020-07-24:
@jhsvip FakeSMC resulted in the same errors, failed to load one ACPI table and then did not load any boot kexts then failed to load NVME driver.  Did you switch to OpenCore?

jhsvip, 2020-07-25:
> @jhsvip FakeSMC resulted in the same errors, failed to load one ACPI table and then did not load any boot kexts then failed to load NVME driver. Did you switch to OpenCore?

I suggest to use vmware to update the beta3.

norton287, 2020-07-25:
@jhsvip do you have a link to a good guide on using VMWare to install it to the drive?

John

serdeliuk, 2020-07-25:
It seems that the install process is not supported right now, also pre installed bigsur is not supported, out there exists a patched 5119 to be able to boot with latest OcQuirks.efi not the ones provided by 5120

jhsvip, 2020-07-25:
Here is the Vmware guide  only chinese version
http://bbs.pcbeta.com/forum.php?mod=viewthread&tid=1862049

opencore + fakesmc  works fine on beta3 installition
I've used for a while

serdeliuk, 2020-07-25:
> My config worked fine for installing Beta 2 no problem, only had to add an NVRAM variable to actually boot into Big Sur.

Where did you added the  NVRAM variable? Maybe you used OpenCore? Clover does not have such config, also official Clover is not yet able to boot or install BigSur, here on post #1 is linked a zip file which contain the patched 5119 which can boot preinstalled BigSur, however, it is not able to boot the installer https://www.hackintosh-forum.de/forum/thread/49311-macos-11-bigsur-dev-beta-clover-patch/?pageNo=1

norton287, 2020-07-25:
@jhsvip gave me the command to execute from my catalina install:

If you enter the desktop after startup, the screen will be blank. Please do the following:
Enter Mac OS 10.15 or enter the installation mode terminal and execute the following command:
sudo nvram booter-fileset-kernel="%00"
Check whether the assignment is successful:
nvram -p |grep booter-fileset-kernel
booter-fileset-kernel %00

norton287, 2020-07-25:
@jhsvip I used OpenCore to install Beta 3, but I am having difficulty entering the recovery, you know how to fix that?

jhsvip, 2020-07-26:
> @jhsvip I used OpenCore to install Beta 3, but I am having difficulty entering the recovery, you know how to fix that?
did you add some commands in boot arags？

norton287, 2020-07-26:
@jhsvip Here is my boot args

[image]

jhsvip, 2020-07-26:
> @jhsvip Here is my boot args
> 
> [image]

just -lilubetaall 
try csr ：77080000  or E7030000

norton287, 2020-07-26:
@jhsvip Will give that a try right now.

norton287, 2020-07-26:
@jhsvip No luck with those changes, gonna dig around inssanelymac.com some more and see.  There were a couple of posts about it with a fix but I cannot find them at all now.

John

jhsvip, 2020-07-26:
> @jhsvip No luck with those changes, gonna dig around inssanelymac.com some more and see. There were a couple of posts about it with a fix but I cannot find them at all now.
> 
> John

my OC efi here https://github.com/jhsvip/ASROCK-Z370M-Pro4-BigSur/releases/

norton287, 2020-07-26:
@jhsvip Fixed it! Now I can enter Big Sur Recovery

[image]

norton287, 2020-07-26:
@jhsvip Thanks for the efi I will clean up mine with yours

jhsvip, 2020-07-26:
> @jhsvip Thanks for the efi I will clean up mine with yours
If you use vsmc to boot 11.0 beta3
remember to add boot arags ：-vsmcgen=1

norton287, 2020-07-26:
@jhsvip I'm using FakeSMC since that boots fine with BS 3

norton287, 2020-07-26:
Closing this as I have Beta 3 installed now.

closed 2020-07-26 by norton287