CloverHackyColor/CloverBootloader #213: r5120 screws up CPU power management on Ivy Bridge laptop

RV-ABZ, 2020-07-25:
After updating my Dell Latitude E6230 from r5119 to r5120, I was no longer able to boot Mojave or Catalina. After much fiddling with CPU PM-related parameters, I was able to boot via single-user mode and recover. However, despite reverting from XCPM setup to legacy CPU-specific SSDT + AICPUPM patch, I still encountered problem booting macOS.

I had to revert to r5119 to retrieve normal booting of either OS, wether with XCPM or CPU-specific SSDT + AICPUPM.

Looks like there is more than OC Quirks support added to r5120.

SergeySlice, 2020-08-28:
My Dell Latitude E6430 booted by Clover 5120 without problem.
If you want to speak about an issue then provide some information, some proof, some logs.

RV-ABZ, 2020-08-28:
:-( 
It's very simple and as I describe...

E6230 with i7-3540M is able to boot with r5110 (off Dids' repo) with either AppleIntelPCUPM patch + generated CPU-specific  SSDT placed in ACPI/patched or KernelPm + XCPM patches + no generated CPU-specific SSDT.

E6230 cannot boot with such settings with r5120. I can only boot through single-use mode. I'm not the only one to complain about trouble with vision r5120/r5121 but if you want something, please be specific. I cannot guess what you want or need for further troubleshooting, if you're at all interested. And I guess you're not given the tone of your reply...

TonyC5, 2020-09-16:
Herve - I'm running CLOVER r5122 on a Lenovo Thinkpad Edge E431 with Ivy Bridge i3-3120M.  My patched ACPI includes an SSDT generated by ssdtPRGen.  I didn't notice any difference when upgrading from r5119 to r5122.  I haven't tried r5120 or r5121.  My DSDT is manually patched (not using CLOVER hot-patches) since I had to patch 8-bit EC anyway.

SergeySlice, 2020-09-19:
RV-ABZ,
you didn't provide any log, any proof, not tell about new Clover version 5122 so I closed the issue as user mistake.

closed 2020-09-19 by SergeySlice