CloverHackyColor/CloverBootloader #218: Problems with dual boot of Windows and MacOS

JamesIslan, 2020-08-07:
Hello. I'm having a problem using dual boot between MacOS and Windows by Clover Bootloader, the problem refers to the Windows audio driver. Basically, when I start Windows 10 using the Clover Bootloader, the system has errors to reproduce the audio, demonstrating a problem in the Device Manager regarding "Intel Smart Sound OED technology". I believe that the problem is not in the construction of my EFI, as I have already checked all the kexts and concluded that there was no conflict with any. I hope this problem is resolved. Att.

SergeySlice, 2021-03-07:
AFAIK the problem is related to AudioDxe driver. 
Exclude it to see if all became OK.
The issue is valid. We have to correct AudioDxe.efi.

SergeySlice, 2021-03-29:
Let join this issue with #187

closed 2021-03-29 by SergeySlice

antoniomcr96, 2021-06-22:
Any news about it? AudioDxe breaks windows audio in many cases, is there any solution?