CloverHackyColor/CloverBootloader #222: Clover_r5120 Graphic boot issue

robaczek66, 2020-08-15:
Hi everyone, any idea:? After new Clover 5120 boot loader instal my booting screen come to be strange. First there is fast text scrolling which before was not seen. After apple logo appear in strange resolution. After some seconds resolution come to normal and MacOs login screen is also normal plus MacOs 10.15.6 seems to work ok with all components. Until Clover 5107 booting was smooth without any text and with proper resolution from beginning. I attached screen photos to visual what happened.
[image]
[image]
[image]
[image]

FrEaK-git, 2020-08-15:
The hell... clean your screen... ;-)

Anyways... which hardware do you use? May it is related to CSM "on" or "off".

Hackintosh, 2020-08-19:
I have a similar issue when trying latest release by building, i get screen but no icons or volumes available
maximus ranger viii

BenTalagan, 2020-08-20:
> Hi everyone, any idea:? After new Clover 5120 boot loader instal my booting screen come to be strange. First there is fast text scrolling which before was not seen.

You might have accidentally installed the CsmVideoDxe driver when updating. You can try to boot without it.

SergeySlice, 2020-09-02:
Never saw such screens. Is the issue still valid with 5122?

robaczek66, 2020-09-03:
Yes I just updated to 5122 and its the same. I use this plist from years, it was used before clover using chimera. It works with all update till 5107 ( I only patched audio many times) Maybe I need to clean everything but not sure what I need from the past drivers

robaczek66, 2020-09-03:
I found my old legacy bootlog and I compare it with current bootlog. I found difference, but why I don't know maybe someone can look at that, please.
First bootlog when resolution was ok from begining and then current one when resolution during beet is not correct
[image]
[image]

SergeySlice, 2020-09-19:
It was 1920x1200 and second screen is 1920x1080. There is the difference.

closed 2020-09-19 by SergeySlice