CloverHackyColor/CloverBootloader #223: CloverV2-5120 build fails when ENABLE_SECURE_BOOT is set

Melab, 2020-08-18:
Attempting to build the source code found in [this](https://github.com/CloverHackyColor/CloverBootloader/releases/download/5120/CloverV2-5120.zip) release file by running `./ebuild.sh -D ENABLE_SECURE_BOOT` fails. The first build log is from a run using unmodified source code. The second is from a run after inserting the line `#include "secureboot.h"` after line 40 in file "${SOURCE_CODE_ROOT}/rEFIt_UEFI/entry_scan/secureboot.cpp`.

[build_log_unmodified.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5091692/build_log_unmodified.txt)
[build_log_after_adding_header.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5091701/build_log_after_adding_header.txt)

Melab, 2020-10-28:
Is there any information on this?

rhaym-tech, 2020-11-25:
Open terminal then type this command "sudo mount -uw /"

Melab, 2020-12-04:
@rhaym-tech

I am running Linux. My `mount` does not have those options. Furthermore, that's not a solution to my problem.

SergeySlice, 2020-12-09:
I can't help you in Linux. The build script is intended for macOS.

Melab, 2020-12-14:
@SergeySlice Does this compilation error not happen when using the build script on macOS? Because as long as I don't specify `-D ENABLE_SECURE_BOOT`, Clover compiles successfully.

jief666, 2021-04-20:
I think the code for secure boot is unmaintained for quite a while. I am personally not sure of what is secure boot. Make sure that efi file are not modified, is that right ?
@Melab do you need this ?
@SergeySlice do we maintain this ?

SergeySlice, 2021-07-06:
There are many errors trying to compile SECURE_BOOT...

jief666, 2021-07-06:
Hence the question : do we support that ? This secureboot code doesn't work and it was never update since the move to C++ et the refactor of menus.
Is there any interest to restore that feature ?

SergeySlice, 2021-07-06:
It was apianti's idea and realisation. I personally have no interest.
Moreover it is UEFI Secure Boot while OpenCore works with Apple Secure Boot which is not the same.

jief666, 2021-07-06:
Melab didn't answer,
Nobody cares about secureboot. The code is anyway not working.
Let's not enable secure boot.

closed 2021-07-06 by jief666

Melab, 2021-07-08:
> I think the code for secure boot is unmaintained for quite a while. I am personally not sure of what is secure boot. Make sure that efi file are not modified, is that right ?
> @Melab do you need this ?
> @SergeySlice do we maintain this ?

Yes, @jief666. I was planning on using it in conjunction with eMMC write protection to create a root of trust.