CloverHackyColor/CloverBootloader #224: 5120 scan

Andrey057835, 2020-08-19:


Andrey057835, 2020-08-19:
Help Please

asgmeonerandom, 2020-08-19:
what are you trying to install? where it fails? add screenshot!

Andrey057835, 2020-08-20:
> what are you trying to install? where it fails? add screenshot!

 I tried instal catalina and high sierra(clover), on different version of clover the same error #
[image]

bfitzgit23, 2020-08-20:
I’m getting stuck here using Big Sur beta 5, no idea why nor how to Troubleshoot and fix it

asgmeonerandom, 2020-08-20:
Paste what he says here https://github.com/CloverHackyColor/CloverBootloader/issues/8#issuecomment-544355811

Andrey057835, 2020-08-20:
> Paste what he says here [#8 (comment)](https://github.com/CloverHackyColor/CloverBootloader/issues/8#issuecomment-544355811)

did you mean debug? in first post it

asgmeonerandom, 2020-08-20:
> > Paste what he says here [#8 (comment)](https://github.com/CloverHackyColor/CloverBootloader/issues/8#issuecomment-544355811)
> 
> did you mean debug? in first post it

yep

Andrey057835, 2020-08-21:
> > > Paste what he says here [#8 (comment)](https://github.com/CloverHackyColor/CloverBootloader/issues/8#issuecomment-544355811)
> > 
> > 
> > did you mean debug? in first post it
> 
> yep

[debug (1).log](https://github.com/CloverHackyColor/CloverBootloader/files/5106694/debug.1.log)

asgmeonerandom, 2020-08-21:
> > > > Paste what he says here [#8 (comment)](https://github.com/CloverHackyColor/CloverBootloader/issues/8#issuecomment-544355811)
> > > 
> > > 
> > > did you mean debug? in first post it
> > 
> > 
> > yep
> 
> [debug (1).log](https://github.com/CloverHackyColor/CloverBootloader/files/5106694/debug.1.log)

your last message is "No AppleSMC Controller" , you should google or post in forum like insanelymac for support.

Andrey057835, 2020-08-21:
> your last message is "No AppleSMC Controller" , you should google or post in forum like insanelymac for support.

Ok, thanks

SergeySlice, 2020-09-02:
Resolved in 5122.

closed 2020-09-02 by SergeySlice

Andrey057835, 2020-09-04:
> Resolved in 5122.

Not resolved,  i install macos on 5120, and then many times it works but one time it show me again scan entries when i update clover to 5121, i tried load my working 5120 clover but nothing changed

Andrey057835, 2020-09-04:
Resolved problem, but here is New
[image]

SergeySlice, 2021-09-25:
New problem - new ticket.
Don't forget to show config.plist and debug.log