CloverHackyColor/CloverBootloader #228: Does Clover Support Big Sur Kext Injection?

Hackintosh, 2020-08-23:
Like in title Does Clover Support Big Sur Kext Injection?

vectorsigma72, 2020-08-26:
not yet for the installer which uses the BootKernelExtensions.kc which is different from the prelinkedkernel

Hackintosh, 2020-08-28:
Thanks, Any ETA Time Frame for the update to support BootKernelExtensions.kc?

SergeySlice, 2020-08-29:
Infinite.

Ubsefor, 2020-09-02:
You could try to use _prelinkedkernel_ in the installer instead of _BootKernelExtensions.kc_

There is not a tutorial, just a remark, as the installer actually seems to support _prelinkedkernel_, but actual kexts are seem to be missing
You can change boot options, because it is still present also there at

/Volumes/<USB INSTALLER VOL>/System/Library/PrelinkedKernels/prelinkedkernel

so we should edit /Volumes/<USB INSTALLER VOL>/Library/Preferences/SystemConfiguration/com.apple.Boot.plist by adding to its
`<key>Kernel Cache</key> <string>System\Library\PrelinkedKernels\prelinkedkernel</string>` (so basically use Xcode or some other plist editors to add or edit Kernel Cache key to supply it with _prelinkedkernel_ location instead of _BootKernelExtensions.kc_ 
This _should_ override the default BaseSystem.dmg (ramdisk) _BootKernelExtensions.kc_ but this requires more testing, probably the `/System/Library/Extensions/` folder should be added to the USB installer for kext matching or BaseSystem.dmg restored to a volume. 

Also it may be necessary to edit Preboot volume boot options with the same ideas later for clover to boot BigSur more easily.

This thing was found by [jackluke](https://forums.macrumors.com/members/jackluke.1133911/);
Link to OP on Macrumors: [link](https://forums.macrumors.com/threads/macos-11-big-sur-on-unsupported-macs-thread.2242172/page-35?post=28610988#post-28610988);

Maybe this will give some ideas to Clover Team!

alessaba, 2020-09-03:
This has probably already been asked, but is it "safe" to upgrade to BigSur with the latest Clover revision, booter-fileset-kernel="%00" variable set coming from Catalina? I don't need to boot into recovery or anything, just upgrade from a completely stable Catalina install.

Hackintosh, 2020-09-04:
I tried to boot bigsur from Clover after Installing it with OpenCore, it did not inject the kexts and clover doesn’t support prelinkedkernel of BigSur yet.

alessaba, 2020-09-25:
I tried booting an existing BigSur installation and all kexts seem to load but I get stuck at 
`busy timeout, (60s): AppleACPICPU`

is this a known problem or just something probably wrong with my config? Catalina works flawlessly

SergeySlice, 2020-10-16:
Resolved

closed 2020-10-16 by SergeySlice