CloverHackyColor/CloverBootloader #229: Somehow disabled AppleALC for RealTek ACL887

laxamar, 2020-08-24:
Version 5121 Somehow disabled Audio (In this case RealTek ACL887). Tried a bunch of layout-ids until I reverted back to 5120 and immediate sound.

Please let me know how I can help to debug

Hackintosh, 2020-08-25:
I have the same issue ALC1150

hla63, 2020-08-25:
Same with alc235 :(

alainjroy, 2020-08-30:
same with alc293 on latitude E7270

laxamar, 2020-08-30:
> same with alc293 on latitude E7270 the message being : alc: @ failed to obtain device info for analog controller (1)

Where did you find the error message?

SergeySlice, 2020-08-30:
Check please with latest git version.

laxamar, 2020-08-31:
> Check please with latest git version.

Not sure what toolchain to install to compile from git source. I have Xcode and access to Linux systems, although I would prefer directly on the Hackintosh. Please point me to the right docs to install toolchain and compile

alainjroy, 2020-09-01:
> > same with alc293 on latitude E7270 the message being : alc: @ failed to obtain device info for analog controller (1)
> 
> Where did you find the error message?

With Hackintool, logs, system logs, search for applealc

SergeySlice, 2020-09-01:
> > Check please with latest git version.
> 
> Not sure what toolchain to install to compile from git source. I have Xcode and access to Linux systems, although I would prefer directly on the Hackintosh. Please point me to the right docs to install toolchain and compile

git clone
./buildme
by default it download and install gcc10
but instead you can use xcode up to 11 version (any case XCODE8)
./buildme XCODE8

laxamar, 2020-09-01:
> git clone
> ./buildme
> by default it download and install gcc10
> but instead you can use xcode up to 11 version (any case XCODE8)
> ./buildme XCODE8

I am getting this:
`BIN_DIR='.' make -C Pccts/dlg
cc -O -I. -I../support/set -I../h -DUSER_ZZSYN -DZZLEXBUFSIZE=65536 -c dlg_p.c
In file included from dlg_p.c:14:
In file included from ../h/pcctscfg.h:61:
../h/pccts_stdio.h:7:10: fatal error: 'stdio.h' file not found
#include <stdio.h>
         ^~~~~~~~~
1 error generated.
make[3]: *** [dlg_p.o] Error 1
make[2]: *** [Pccts/dlg/dlg] Error 2
make[1]: *** [VfrCompile/VfrLexer.h] Error 2
make: *** [Source/C] Error 2`

SergeySlice, 2020-09-02:
OK, test release 5122.

laxamar, 2020-09-02:
Release 5122 has fixed this issue. Will test other features, but closing this one

closed 2020-09-02 by laxamar