CloverHackyColor/CloverBootloader #230: Unable to boot after upgrade to v5.0 r5121

liyafe1997, 2020-08-27:
Everything works well before (r5120 and pervious version).
I just replace CLOVERX64.efi with r5121 one and reboot my computer. And a error message shown as the picture.
[image]
Now I replace back with r5120's CLOVERX64.efi and everything back to me.

Airtonjunior54, 2020-08-27:
Same problem here. Needed to go back to r2120

Sknny, 2020-08-28:
Same. For me problem was caused by BOOTX64.efi

chriseg93, 2020-08-28:
I had the same exact issue, I found that if you upgrade to r5121, but then use the BOOTX64.efi file from r5120, it will boot, but it will still show as r5120 when booting.

SergeySlice, 2020-08-29:
> I had the same exact issue, I found that if you upgrade to r5121, but then use the BOOTX64.efi file from r5120, it will boot, but it will still show as r5120 when booting.

Nonsense. If you boot by BOOTX64.EFI then you boot to Clover 5120 which is.

chriseg93, 2020-08-29:
> > I had the same exact issue, I found that if you upgrade to r5121, but then use the BOOTX64.efi file from r5120, it will boot, but it will still show as r5120 when booting.
> 
> Nonsense. If you boot by BOOTX64.EFI then you boot to Clover 5120 which is.

I never said it would be r5121, I just stated what I did and that it showed up as 5120, I don't know anything beyond that.

Najip, 2020-08-30:
Same problem

Javichuuu, 2020-08-30:
Same problem

SergeySlice, 2020-08-30:
Confirm please that the issue resolved by 493aa8

chriseg93, 2020-08-30:
@SergeySlice How would we go about editing 493aa8 into the system?

liyafe1997, 2020-08-31:
> 
> 
> @SergeySlice How would we go about editing 493aa8 into the system?

You need to compile it by yourself. Or just waiting for the release.

chriseg93, 2020-08-31:
> > @SergeySlice How would we go about editing 493aa8 into the system?
> 
> You need to compile it by yourself. Or just waiting for the release.

Thank you very much, I have no clue how to do that, guess we'll just have to wait for the next release, thank you!!

chriseg93, 2020-09-02:
@SergeySlice I just tried r5122, it now boots up fine, but when I select macOS to boot up, it throws the same NULL error.

LeeBinder, 2020-09-02:
@SergeySlice pretty much same thing here as for chriseg93. I'm still in the process of narrowing down config.plist entries and drivers/UEFI. Will report back either with more info or with a new debug.log.zip soon.

LeeBinder, 2020-09-02:
@chriseg93 can you boot into GUI and then into macOS with config.plist removed or renamed to e.g. config.#plist? If so, next try rename it back, open config.plist in a text editor and rename `<key>GUI</key>` into e.g. `<key>#GUI</key>`, and `<key>ForceKextsToLoad</key>` into e.g. `<key>#ForceKextsToLoad</key>`

save, reboot, report back please.

LeeBinder, 2020-09-02:
I can only boot into Clover 5122 GUI after removing ApfsDriverLoader.efi from drivers/UEFI, which beats the purpose because then I can not boot into my Mojave and Catalina APFS volumes. Yes, I let Clover_r5122.pkg update any of my UEFI drivers, and afterwards also double-checked they were all updated. Booting into my HFS+ High Sierra volume (with HFSPlus.efi present in drivers/UEFI) works fine with 5122.

Same with the previous version of ApfsDriverLoader.efi

@chriseg93 if you can boot into Clover 5122 GUI with an ApfsDriverLoader.efi present in drivers/UEFI, could you please zip up your entire drivers/UEFI folder and attach it here? Thank you.

@SergeySlice the problem with ApfsDriverLoader.efi present, Clover does not even generate a debug.log in misc, so I can't even attach any..

I also tried booting w/o any config.plist - same issue.

LeeBinder, 2020-09-02:
@SergeySlice I just let my notebook sit for a long while and eventually Clover _**did**_ generate a debug.log! I uploaded it into my comment above - here the same link for your convenience: [debug.log + config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5165071/debug.log.%2B.config.plist.zip)

chriseg93, 2020-09-02:
Yea, no luck, just ended up going back to r5120

LeeBinder, 2020-09-04:
closer to a final fix, see https://github.com/CloverHackyColor/CloverBootloader/issues/226#issuecomment-687063418

LeeBinder, 2020-09-04:
for me, [solved](https://github.com/CloverHackyColor/CloverBootloader/issues/226#issuecomment-687094658) thanks to jief666's fixes!

LeeBinder, 2020-09-08:
@liyafe1997 please close this issue - thank you.

closed 2020-09-08 by liyafe1997