CloverHackyColor/CloverBootloader #231: Editing boot args on the fly does not work

laszlo-t, 2020-08-28:
When I edit the boot args in the Clover menu, I get kernel panic. But when I use the same boot args in the config file, it works.

I'm using Clover v5.0 r5121 on Catalina USB installer.

My steps:

1. the config.plist contains the following boot args:
`<string>-v -no_compat_check npci=0x2000 dart=0 debug=0x100 keepsyms=1</string>`
2. add `-s` next to `-v` in the Clover menu and boot
3. kernel panic
4. try again with adding `-s` in the config.plist and do not change anything else
`<string>-v -s -no_compat_check npci=0x2000 dart=0 debug=0x100 keepsyms=1</string>`
5. it works with the same boot args, the single user mode is running

I attached the [config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5143498/config.plist.txt) and some pictures. You can see the boot args are the same but somehow it does not work when I changed in the Clover menu.
[image]
[image]
[image]

jief666, 2020-08-30:
Hi,
I've just tried and it worked for me.
Could you try with this efi [CloverX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5147385/CloverX64.efi.zip) and tell me if the issue still there ?
Thanks.

laszlo-t, 2020-08-30:
Hello,
Thanks for you response.

I tried that version and also played the order of boot args but unfortunately it still does not work.
Here is a slightly modified [config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5147555/config.plist.txt)

I enabled the debug mode and attached some logs:
This is where I added the `-s` in the clover menu and got kernel panic:
[debug1.log](https://github.com/CloverHackyColor/CloverBootloader/files/5147559/debug1.log)

and here I set in in the config file and reached the single user mode:
[debug2.log](https://github.com/CloverHackyColor/CloverBootloader/files/5147560/debug2.log)

I compared the two files and see some disturbing changes. I'd like to highlight again I only changed the boot args with `-s` in the above examples.
(Hint: an ugly regexp to remove the timestamps: `^\d+:\d+  \d:\d+  `)

jief666, 2020-08-31:
Your logs allow me to catch it. Build the latest commit from github or try this : 
[CLOVERX64.efi.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5150242/CLOVERX64.efi.zip)

Please confirm it works for you, and close the issue if it is.
Thanks.

laszlo-t, 2020-08-31:
I tried your build and did some tests.
It works great, no issues.
Thank you.

closed 2020-08-31 by laszlo-t