CloverHackyColor/CloverBootloader #233: Clover_r5114-Clover_r5122Graphic boot issue

jackacc, 2020-09-03:
R5114 or below supports my computer 10.13.6 or below. It has been changed from 10.10 to 10.13 with good clover. 10.14 can also be patched with sse4.2 and N card. Both can run. After the r5115 version is updated, my computer is no longer supported, and the N card gt320 cannot be turned on. It is suggested that the author still use the previous code to find a breakthrough and be compatible with the latest 11

jackacc, 2020-09-13:
[image]
Old kernel not supported?

SergeySlice, 2020-09-19:
Kernel from 10.7 is still supported.
Search for own mistakes.

closed 2020-10-03 by SergeySlice