CloverHackyColor/CloverBootloader #234: Clover 5122 Stuck on Scan Entries

Hackintosh, 2020-09-03:
Maximus Ranger VIII

Hackintosh, 2020-09-03:
It was a theme issue, something related to old theme.

arnaudpladys, 2020-09-04:
Just happened to me, any solution ?

Hackintosh, 2020-09-04:
Yes, remove the old theme and theme entry from config.plist. This solved my issue, it might work for you.

Hackintosh, 2020-09-04:
Yes, remove the old theme and theme entry from config.plist. This solved my issue, it might work for you. You don’t need to add another theme, clover has built in theme.

arnaudpladys, 2020-09-04:
Hum! Silly question, you need to create a bootable usb to acces it? Because at the moment the only bootloader i have is clover and of course it’s not loading? What is the easiest way to edit the config.plist?

Hackintosh, 2020-09-04:
Just use BDUtility it should work, i still have no idea of editing config.plist from windows or linux.

arnaudpladys, 2020-09-04:
Humm... this is not working. I remove the theme from my config.plist, same, still stuck

arnaudpladys, 2020-09-04:
I rolled back from my backup on 5120 using a ubuntu bootable usb.. It's back up again. Not going to upgrade clover yet ahah!

originalmagneto, 2020-09-06:
> It was a theme issue, something related to old theme.

This did not solve this issue, having the same one on my config. Must be another issue, not related to themes man :/

ifamed, 2020-09-07:
I have the same problem, the clover boot options don't even appear on z270x-ud3 (r5122/r5121).
Version r5120 works perfect.

SergeySlice, 2020-09-19:
If a Clover stuck at scan entries then you should provide debug.log.
Else the issue will be closed as wrong.

GeorgeJames23, 2020-09-22:
I have created an account so I could reply to this response. I too, as it would seem many others, had the problem of being stuck on “Scan Entries” . Just because debug.log  is not forwarded does not mean the problem didn’t exist. I much appreciate your efforts in creating etc Clover Bootloader otherwise my computer would not run but to get back to a boot loader that worked (5120)  from 5121 nearly ended in distaster. I would be grateful if you’d look into this issue and restore confidence that we can safely use clover bootloader updates.

SergeySlice, 2020-10-03:
There is no issue. There is somehow user mistake that can be resolved by looking to debug.log.
10days no reply = no issue.

closed 2020-10-03 by SergeySlice