CloverHackyColor/CloverBootloader #235: The button of Clover Boot Option does not work, Would you

vincentchen666, 2020-09-03:


ifamed, 2020-09-07:
I have the same problem, the clover boot options don't even appear

LeeBinder, 2020-09-07:
does the O key still work?

LeeBinder, 2020-09-09:
all buttons in GUI incl. Options working fine here with the latest updated 5122 release, both click onto button and also via O key

vincentchen666, 2020-09-11:
@LeeBinder  The option of add Clover boot options  for all entries  does not work.   I could not add Clover bootloader menu into the boot menu of motherboard.

LeeBinder, 2020-09-11:
oh boy. I doubt this is only a language issue... If you want help with this, please change the title of this issue of yours into

**Need help re-adding Clover boot entry to BIOS boot menu**

Thank you.

oldnapalm, 2020-09-28:
I have a similar issue with Asus Prime B250M-Plus, the menu "Clover Boot Options" is there but "Add Clover boot options for all entries" does not work.
If I enter shell and run `bcfg boot add 0 cloverx64.efi "Clover"` from EFI/CLOVER it works.

SergeySlice, 2020-10-16:
It works

closed 2021-02-21 by SergeySlice