CloverHackyColor/CloverBootloader #236: Unable to create bootable usb in winows10

GizmoTheGreen, 2020-09-04:
tried makeusb.bat for some reason is creates usbs32.bin instead of usbbs32.bin like spelled in the .bat??

also at the step 
@Genbootsector.exe -o %EFI_BOOT_DISK% -i boot1f32.bin
(also tried manually Genbootsector.exe -o e: -i boot1f32.bin)
fails with 
ERROR: Write DBR: failed - Failed to read/write files (LastError: 0x5)!

the other commands in the bat all finish cleanly/with success.

my goal is to create a bootable device that will chainload (hopefully) windows 10 off of a PCIE ssd on legacy bios computer. just for kicks.

Already booted EFI windows install from sata drive with bootable clover CD.

thanks.

SergeySlice, 2020-09-19:
Best wishes!
The makeusb.bat is too old to discuss. We are using other methods.

GizmoTheGreen, 2020-09-19:
There is no tutorial for other methods. Maybe include a txt with guide and what program to use? I also tried the commands for linux (on arch linux) still didn't work.
It's also not clear that you must have boot file, and what boot file to copy and what name to give it.
I had to puzzle it all together by deducing through reading scripts and comments. None of them match up with the new versions. So why is so old files even included...

SergeySlice, 2020-10-03:
I know how to do this in macOS and Linux but Windows is not my field of interest.

closed 2020-10-03 by SergeySlice