CloverHackyColor/CloverBootloader #237: Clover beta testing guidelines

LeeBinder, 2020-09-07:
Hello Clover devs - would you please share how your beta testing guidelines are structured? Like:

1. who all beta tests before public release - how many people are in that group?
2. how many days are allotted for beta testing once the dev team agrees on that alpha (active coding) state has ended and beta state begins?
3. how many beta channel participants' "thumbs up" are required before final public release?
4. on what platform (gitter, telegram etc.) does beta testing occur?

In light of frequent bugs and Clover not booting anymore esp. during the last few months, insights into the internals (and hopefully some updated and streamlined approaches) would be greatly appreciated.

stevezhengshiqi, 2020-09-08:
I believe it's more like a volunteer-to-test way for the most open-source projects. Different from commercial or close-source programs, Clover reveal all its code and everyone can just clone and generate the beta version on their own by running [buildme](https://github.com/CloverHackyColor/CloverBootloader/blob/master/buildme). If people find bugs, they can just report it here on GitHub Issue page.

At this stage, Clover team is busying integrating OpenCore libraries to support Kernel Collection mechanism in Big Sur. For anyone who has interest, https://github.com/CloverHackyColor/CloverBootloader/tree/opencore_integration is a good start.

BTW, there're way too many "requesters" than developers for Clover. I believe Clover team would be happy to see if people can dig down into their problems and send pull requests to improve this bootloader.

LeeBinder, 2020-09-08:
thanks Steve for giving some background. 

I think the decision to run a beta channel is independent of a open-source or commercial aspect. Two open-source apps with pre-release betas that comes to my mind right away are the Signal app for a variety of platforms, and IINA (both also on github).

Also @SergeySlice @jief666  Exactly because of the current phase of integrating OpenCore libraries (which obviously self-defines as beta phase) and other code restructuring, it appears sensible to at least **label each release beta for now** to begin with, so people know what they might get themselves into; then observe the issues section for lets say seven days, and **only when all is clear** (either w/o any more surgery, or if required after fixing of severe issues caused by the current update), then push the final.

Please consider honestly if this can be the safest route for now for all, you devs and the users. And yes, I'm 100% positive there will be plenty people who can't wait to even install Clover if it's marked beta. I dare say more sensitive approach will be much less embarrassing than what's been happening with each of the three latest "releases" 5120/5121/5122 which all were not final when they were pushed but had to be (and silently were) replaced with the respective final commits on the release page.

And more than that, there is not disadvantage at all of being careful and adding a beta-label, esp. during this transition - it's just being honest. Thanks for discussing.

btwise, 2020-09-09:
A new generation of **Open-Clover** is coming soon!

stevezhengshiqi, 2020-09-09:
Just a careful reminder that Clover is not a commercial product, so the developer team receive no profits from maintaining the project. It’s totally dependent on their self-interest that sustain the whole bootloader. Rather than complaining the development stage, I prefer people to do “real stuff” to help Clover team in technical way or financial way.

socialskyo, 2020-09-09:
@stevezhengshiqi Totally agree.

asgmeonerandom, 2020-09-09:
> Just a careful reminder that Clover is not a commercial product, so the developer team receive no profits from maintaining the project. It’s totally dependent on their self-interest that sustain the [...]

I don't think you understand the issue, as he claimed he would like to know if there is a way to join Clover discussions and as well as development.
Many folks here have programming experience but when they see the project they get shocked and don't know how to start.

jief666, 2020-09-09:
I agree, it's not easy to start. I felt that too. I can help programmer to dive in and join.

asgmeonerandom, 2020-09-09:
> I agree, it's not easy to start. I felt that too. I can help programmer to dive in and join.

yes, how can we contact you?

LeeBinder, 2020-09-09:
@stevezhengshiqi you misinterpret: I am not complaining but a) merely analyzing the current situation by stating facts then b) mentioning possible solutions. Again, if profit or non-profit or who's contributing to the project and who's not is totally irrelevant for issuing a compiled commit either in the appropriate channel or at least with the adequate label, in the cases of all recent ones BETA. It's much rather about facing reality which is new bugs introduced with each update, some of which severe, keeping Clover from booting. That's they way it is and totally fine during work-in-progress, but it should really be tagged accordingly, at least on the release page here on github, in the file name (and best also in the installer) along with a warning "BETA - please report new bugs at https://github.com/CloverHackyColor/CloverBootloader/issues incl. this BETA release build #". That's all.

Honestly, I don't see *any* reason why not to lean this way. The other option to continue treading on the old worn path is worse because a) it will shock some users away who feel they did not opt into beta testing, and b) keeps on putting you devs in the cumbersome position to - issue over issue - explaining perpetually what happened.

Let me know if anything is still unclear and I'll try to lay out the plan some more.

@asgmeonerandom interesting what you read into my words. If you consider my detailed bug and beta testing reports of my shared troubleshooting adventures sufficient to count as "Clover development" - fair enough (even though to me that's too much of a 🥇  .. ;) )

stevezhengshiqi, 2020-09-09:
>  Again, if profit or non-profit or who's contributing to the project and who's not is totally irrelevant for issuing a compiled commit either in the appropriate channel or at least with the adequate label, in the cases of all recent ones BETA

Do you realize that you are asking a favor from a non-profit development group who has no responsibility to do so? It totally depends on their free will, so it's better not to push them too hard. If you really want to help them, I think you can help create the channel and invite the Clover team. BTW, it's very easy for users to compile on their own: just running the buildme file in the root directory.

LeeBinder, 2020-09-10:
This is not a favor I'm asking for, and neither am I "pushing" anything because simply I can't - I am merely sharing a suggestion based on observation of facts. Sure, no responsibility, and also sure, free will on the side of the dev team. An non-dev outsider opening a new channel w/o prior mutual consent would be jumping the gun. Any such move would need to be initiated by the dev team as a whole.
`The true meaning of the opensource projects is that everyone can help and reduce the workload carried by the main developers.`
"everyone" can only contribute depending on their knowledge, skills etc. In my case w/o sufficient coding skills that's been beta testing, reporting, and eventually sharing my findings by opening this topic.
`BTW, it's very easy for users to compile on their own: just running the buildme file in the root directory.`
And will it happen - is it realistic or rather hypothetical, like "any smoker can just stop smoking"? Will an armada of users suddenly rush to self-compile every commit and test? No. The issue that's been solidifying lately and which I've raised in this topic remains.

Maybe a good time to let everything sink and sleep on it for a night 🔢

jief666, 2020-09-10:
> > I agree, it's not easy to start. I felt that too. I can help programmer to dive in and join.
> 
> yes, how can we contact you?

InsanelyMac, pseudo jief666

jief666, 2021-04-20:
@asgmeonerandom Did you try to contact me about starting development for Clover ?

jief666, 2024-01-05:
Closed because stalled.

closed 2024-01-05 by jief666