CloverHackyColor/CloverBootloader #238: Question about OC Integration and boot options

MysticalOS, 2020-09-10:
I've watched OC develop impressively, but one thing that's always kept me from OC was the fact that it's got an all or nothing approach. It lacks per OS rules/patching. If OC is installed you basically have a mac, whether you boot windows or macOS or something else, it applies it's modifications to make your machine a "Mac". I've always appreciated that clover allows you to specify OS patching and such so if you boot windows via clover, it's still just a PC, not a PC pretending to be a mac booting into windows.

So my question here, when integrating OC libraries and patching, will clover team ensure that this level of customizing remains present?

SavageAUS, 2020-09-14:
+1 for the above. I would like for Clover to remain as is once opencore is integrated.

stevezhengshiqi, 2020-09-14:
@MysticalOS Actually, you just need some settings to avoid OpenCore from changing your model in Windows. 
`PlatformInfo - UpdateSMBIOSMode` -> `Custom`
`PlatformInfo - SpoofVendor` -> `false`
`Kernel - Quirks - CustomSMBIOSGuid` -> true

SergeySlice, 2020-09-15:
The answer is Yes. Clover will remain Clover and it will boot Windows as PC not as Mac.

closed 2020-09-15 by SergeySlice