CloverHackyColor/CloverBootloader #239: NVRAM Emulation

VisionXi, 2020-09-11:
Did something in NVRAM emulation change ? With versions < 5120 it worked well, but with 5122 the clover menu doesn't remember the last picked boot item.

SergeySlice, 2020-09-19:
Not true.

VisionXi, 2020-09-20:
What an arrogant answer! You can believe a guy with over 30 years IT experience.
I can reproduce it, vice versa and every time. If you can not then there are other differences in configuration which affect this.

asgmeonerandom, 2020-09-20:
> What an arrogant answer! You can trust a guy with over 30 years IT experience.
> I can reproduce it, vice versa and every time. If you can not then there are other differences in configuration which have an impact on this.

30 years experience and you can't afford a mac?

VisionXi, 2020-09-22:
> > What an arrogant answer! You can trust a guy with over 30 years IT experience.
> > I can reproduce it, vice versa and every time. If you can not then there are other differences in configuration which have an impact on this.
> 
> 30 years experience and you can't afford a mac?

It costs a lot if you have kids and support them so that they can write funny notes in the net sometime. Have fun. Goodbye!

SergeySlice, 2020-10-03:
With Clover 5122 the menu pointed to last booted entry. As well as with all older revisions as never changed.

closed 2020-10-03 by SergeySlice