CloverHackyColor/CloverBootloader #240: Since v5122: XBuffer::[] : i > _Len. System Halted

chatelp, 2020-09-15:
Hi ! Since upgrading clover from v5120 to v5122, I get the following error when booting Catalina (10.15.6). Reverting back to v5120 it works again: so exact same configuration, but fatal error at boot time with v5122 only.

I tried googling this XBuffer array to no avail.

SergeySlice, 2020-09-19:
Should be resolved with updated release 5122. Try again.

chatelp, 2020-09-19:
> Should be resolved with updated release 5122. Try again.

I already did. Unfortunately, it’s not resolved.

tlefko, 2020-09-23:
@chatelp can confirm

SergeySlice, 2020-10-03:
Provide your config.plist and debug.log to find something related to your problem.

chatelp, 2020-10-03:
Here is my config.plist. I already reverted back to clover v5120 so debug is not relevant. If you need it, I can revert back to 5122/5123.

[config.plist.txt](https://github.com/CloverHackyColor/CloverBootloader/files/5321697/config.plist.txt)

SergeySlice, 2020-10-16:
user mistake

closed 2020-10-16 by SergeySlice

chatelp, 2020-10-16:
@SergeySlice would you be so kind as to give me a little more feedback on why it’s a « user mistake » ? Thanks.

chatelp, 2020-10-30:
@tlefko did you finally find a solution for clover r5122 ?