CloverHackyColor/CloverBootloader #242: Release v5.0 r5122

nyhtml, 2020-09-30:
Release v5.0 r5122 is causing the GPU to display incorrectly on Mojave.
[image]
Replacing the BootX64.efi with CLOVERX64.efi from Release v5.0 r5114 fixes the issue.
[image]

SergeySlice, 2020-10-16:
Recheck please with 5125.

nyhtml, 2020-10-31:
I'll test it now. Thank you.

closed 2020-12-09 by SergeySlice

nyhtml, 2020-12-20:
It didn't help (which is fine) on the **Optiplex 9020 SFF** but I was able to move up to **Clover EFI v2.5k r5116**.
Since Mojave has no more updates coming from Apple, 5116 will be alright going forward on this system.
I still have my Z270XUG-CF that is running Catalina 10.15.7 on **Clover EFI v2.5k 5107** so I may need a SFF GPU.

If not off topic, a compatible GPU would be appreciated.