CloverHackyColor/CloverBootloader #243: Upgrade, Installation Fails

arrmo, 2020-10-02:
My apologies up front! I see this posted several times, but pulling my hair out, as I just can't get it to work 😞. Likely me, but ...

I have tried the different items captured here, no joy, https://github.com/CloverHackyColor/CloverBootloader/issues/6

Also, tried like this, still no joy,
https://hackintosher.com/forums/thread/enable-disable-system-integrity-protection-sip-on-a-hackintosh.53/

In all cases, the installer isn't recognizing the EFI partition that I need to install (upgrade) to - I see others listed, but not that one. And I can edit config.plist, so that partition is accessible and writable.

Any suggestions?

Thanks!

flynn-buc, 2020-10-06:
I have the exact same problem :/

SergeySlice, 2020-10-16:
Security is not a Clover problem.

closed 2020-10-16 by SergeySlice