CloverHackyColor/CloverBootloader #246: [5123] Failed to boot Mojave and Big Sur

stevezhengshiqi, 2020-10-04:
macOS Version: 10.14.6 & 11.0beta8
CPU & GPU: i5-8250U & UHD620
Laptop

Hi, thank you for making the big changes. I have a problem booting Mojave and Big Sur. Here are my verbose log and EFI folder. Really appreciate it if someone could have a look.
[image]
[CLOVER.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5322938/CLOVER.zip)

SergeySlice, 2020-10-16:
use 5125

closed 2020-10-16 by SergeySlice

stevezhengshiqi, 2020-11-05:
https://github.com/CloverHackyColor/CloverBootloader/commit/5d91bbe25305104192160413b68f94955e7cb03a solves the problem, thank you guys.