CloverHackyColor/CloverBootloader #247: Did not find config.plist/quirks. Fatal Error System halted

mruepp, 2020-10-04:
Update from Clover 5022 to 5023. Nothing changed.

mruepp, 2020-10-04:
I would suggest creating a different release path for the oc integration and remove this path from the Clover Update Prefs Pane. Possibility to brick a lot of systems with the current approach

stevezhengshiqi, 2020-10-04:
> Nothing changed.

As the instruction in https://github.com/CloverHackyColor/CloverBootloader/releases/tag/5123 says, there’s new quirks in config.plist. You have to adapt those new entries and update your config.plist.

mruepp, 2020-10-04:
Its just that I suggest to not distributed possibly breaking pre releases through the update mechanism in the Clover prefs pane.

jamiehunt, 2020-10-04:
Anyone with a bricked system because of this here is what I've done to resolve:

1. Boot from a Ubuntu live USB
1. Mount your EFI partition
`sudo fdisk -l`
Identify your efi partition, (probably /dev/sda1)
`sudo mound /dev/sda1 /mnt`
1. Import quirks section into your config.plist file
`sudo vim /mnt/CLOVER/config.plist`
Copy in the "Quirks" key from sample -> https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist#L1088

This has got me back into clover but hanging on Apple logo, investigating further. Maybe you also want to take the opportunity to backup any data from your APFS partition when in Ubuntu.

drashna, 2020-10-04:
> Its just that I suggest to not distribute possibly breaking pre releases through the update mechanism in the Clover prefs pane.

Exactly this.  It shouldn't be pushed out yet. It breaks things, and means you basically need to disable updates checking, outright. Neither is a good look.  To the point that it may be simpler to just dump clover and move to openCore, outright.

1Revenger1, 2020-10-04:
@jamiehunt looks like you have duplicate kexts in your EFI and L/E. If you want to try the update again, try clearing out 3rd party kexts in L/E and invalidating your kernel cache.

eskimo, 2020-10-04:
Like @jamiehunt suggested for bricked systems, you can also fix in windows if you have a dual boot by mounting your EFI like so:

- WIN+R
- Type `cmd.exe` then press enter.
- Type `diskpart`
- Use `list disk` to list then `select disk X` to select
- Use `list partitions` to list then `select partitions X` to select
- Type `assign letter=e` (replace e with any letter you want)

After that you can browse your EFI partition and add the quirks to your config. You will need to open notepad or any other text editor with administrator permissions to be able to open/edit file.

jamiehunt, 2020-10-04:
> @jamiehunt looks like you have duplicate kexts in your EFI and L/E. If you want to try the update again, try clearing out 3rd party kexts in L/E and invalidating your kernel cache.

@1Revenger1 Ah, thanks for the analysis. Yes I had duplicate kexts between 10.15 and 'Other'.

YaarPatandarAA, 2020-10-05:
> EDIT: Rolled back clover and working fine again. 🎉

Could you please tell us how you rolled back clover.

jamiehunt, 2020-10-05:
> Could you please tell us how you rolled back clover.

@YaarPatandarAA Check the install log txt file on your EFI partition, this should say where it copied everything as a backup before the update, if you can get to that backup directory, just copy it back.

Else download a new copy of Clover (at whatever version you are interested in) from the releases tab, download the full zip i.e. CloverV2-5122.zip and copy over the EFI directory from that ZIP to your EFI partition.

mruepp, 2020-10-05:
You can also mount the disk somewhere else and reinstall from clover pkg to the mounted system disk. (you can mount gor instance on a mac) Be sure to choose the right volume. I personally use a emergency installation on a usb stick to boot and fix things.

Neko-Monika, 2020-10-05:
I got the same error where it also said about missing "DisableIoMapper" key in config.plist. In config-sample.plist in "OEM" folder, unfortunately, that key wasn't exist, not to mention "Quirks" dict, so I downloaded latest Clover ISO and copy all quirks in my own config.plist from there. Clover had booted, but before welcome message I saw "NotFound returned from DataHubDxe.efi" and installed macOS Catalina didn't start: it show Stop sign instead of Apple logo. But all drivers, including DataHubDxe, were in "CLOVER/drivers/UEFI" during boot attempt.

So yeah, I roll back Clover's service files in its EFI partition on disk and will decline CloverUpdater's update request for the time being.

shilik, 2020-10-05:
I got the same errors. My solution is the following:

add
<key>DisableIoMapper</key>
<true/>
in the  Quirks section, then remove DataHubDxe.efi from
"CLOVER/drivers/UEFI"

reboot, and it works like charms.

vacarogit, 2020-10-05:
Who did this to duck up my dual boot clover setup? Fuck

vacarogit, 2020-10-05:
@mruepp my high Sierra on clover is deeply fucked up by this stupid release:triumph:

mruepp, 2020-10-05:
> @mruepp my high Sierra on clover is deeply fucked up by this stupid release😤
> 
> To the kind author, please be considerate, not all people embrace opencore, many still use clover with Aptiomemoryfix pack, you don't mess with people's system 😾👎

Why are you tagging me? I am just the reporter because I had this issue as well. I encourage you to calm down and be kind, all the developers here are doing this for free. So fix your system and carry on.

closed 2020-10-05 by mruepp

ivnnv, 2020-10-07:
Well done guys, introducing breaking changes in a regular release without warning. Yeah why not!

Thrashavich, 2020-10-09:
I’m a noob and don’t know what quirks are. Luckily I am able to boot to my windows drive using a unibeast usb with and older clover revision. I came across this problem at 8:50PM. Let’s see how long it takes me to get my system working again.

Thrashavich, 2020-10-09:
> Like @jamiehunt suggested for bricked systems, you can also fix in windows if you have a dual boot by mounting your EFI like so:
>
> * WIN+R
> * Type `cmd.exe` then press enter.
> * Type `diskpart`
> * Use [...]

Can you help me out? The “list partitions” step takes me elsewhere. 

[image]

Thrashavich, 2020-10-09:
[image]

After bricking my system and booting into my windows 10 drive via and unibeast flash drive with an older revision of clover, I am unable to access my EFI partition from windows 10. I cannot boot into OS X. I tried take ownership ex and several other methods of taking ownership. I cannot get permission to open my efi folder. Can anyone help?

1Revenger1, 2020-10-09:
Windows is super protective of the EFI partition - you have to either use explorer++ opened with administrative permissions, restart explorer.exe with admin privileges, or use CMD with admin privileges.

Thrashavich, 2020-10-09:
> Windows is super protective of the EFI partition - you have to either use explorer++ opened with administrative permissions, restart explorer.exe with admin privileges, or use CMD with admin privileges.

You’re a Saint. Thank you for the super quick reply. Downloaded this now. I’m gonna just copy my backup clover files to this partition and hopefully things will be back to normal.

Thrashavich, 2020-10-09:
> Windows is super protective of the EFI partition - you have to either use explorer++ opened with administrative permissions, restart explorer.exe with admin privileges, or use CMD with admin privileges.

Copied my old efi folder to my efi directory and I’m up and running on OS X again thank you so much dude you’re a life saver. I use OS X for work and needed to get something done tonight that I couldn’t accomplish on my old Mac book pro. If you’ve got a Venmo private message me and I’ll send you a case of beer. Thank you so much.