CloverHackyColor/CloverBootloader #248: touchpad I2c network (VoodooI2c not loa) after update to 5123

HandaTakuya, 2020-10-04:
I update clover to 5123 from 5122 after reboot my touchpad not work. I check with GenI2c show VoodoI2c not load!
I try update Kextcache but NOT work

final... I install 5122 again to work

1Revenger1, 2020-10-04:
https://github.com/CloverHackyColor/CloverBootloader/blob/dfac7119f6032c7520c0e3245bd5f914c6f99b15/rEFIt_UEFI/refit/main.cpp#L1315
I'm guessing it's an issue with kext order - VoodooI2c has multiple kexts which have to be loaded in a specific order.

HandaTakuya, 2020-10-05:
> https://github.com/CloverHackyColor/CloverBootloader/blob/dfac7119f6032c7520c0e3245bd5f914c6f99b15/rEFIt_UEFI/refit/main.cpp#L1315
> 
> 
> I'm guessing it's an issue with kext order - VoodooI2c has multiple kexts which have to be loaded in a specific order.

HandaTakuya, 2020-10-05:
> https://github.com/CloverHackyColor/CloverBootloader/blob/dfac7119f6032c7520c0e3245bd5f914c6f99b15/rEFIt_UEFI/refit/main.cpp#L1315
> 
> 
> I'm guessing it's an issue with kext order - VoodooI2c has multiple kexts which have to be loaded in a specific order.

I can't find rEFIt_UEFI/refit/main.cpp in My EFI

closed 2020-10-05 by HandaTakuya

luka-diplomski, 2020-10-11:
How to specify order in Clover?

This should be reopened