CloverHackyColor/CloverBootloader #253: r5123 CANNOT boot.

etajv100, 2020-10-07:
Before upgrading to 5123 from 5122, i updated Lilu and WhateverGreen, like i always do, then updated to r5123, now i cannot boot anymore,

i7 870
Biostar TH55B HD
GTX 680

i saw somewhere some people talking about quirks, but not sure what to do, just copy them to my old config plist or, modify them or what exactly.

for now i'll be downgrading to 5122, until something gets done about this.

Neko-Monika, 2020-10-07:
I think, we all should wait till new stable version's release, because some users also had encountered with non-working VoodooI2C kext.

etajv100, 2020-10-07:
Yeah true, bue i guess the "This version successfully loads BigSur and allows install and update" was just too exiting for me that i didn't notice it was a pre-release.

SergeySlice, 2020-10-16:
Use 5125.
About VoodooI2C there is another issue.

closed 2020-10-16 by SergeySlice