CloverHackyColor/CloverBootloader #256: Can't boot Catalina installer or Big Sur with r5123

zty199, 2020-10-08:
macOS version: 11.0 Beta9 & 10.15.7 installer
CPU: i5-7300HQ
GPU: HD630

Thank you for your contribution to make Clover compatible with Big Sur. I saw some one has booted Big Sur successfully, but unfortunately when I use r5123, it can't boot Catalina installer or Big Sur Beta9 either. It always stucks at somewhere strange.

[image]

At first I thought it was related to EC or RTC problem, but adding relevant SSDTs and DSDT patches didn't work. It still stucks at similar place. Here's my config.plist and debug.log.

[config.plist.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5347137/config.plist.zip)
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/5347138/debug.log)

This config is what I used before, and it works well in r5122, which can boot Catalina 10.15.7 installer.
I simply copied the Quirks part from new config-sample.plist in pre-release 5123 and then added it to mine. And the settings are what I use in OpenCore, which can boot Big Sur successfully. I have tested few versions from pre-release 5123 to latest commit, still can't make it work.

Hope someone can help me find out the reason......

zty199, 2020-10-08:
I thought it was caused by part of 'KernelAndKextPatches' settings in Clover and Quirks in OpenCore is duplicated, so I disabled relevant part in config...... But still nothing changes......

zty199, 2020-10-11:
I think I found something new. I tried to add boot-arg "npci=0x2000" and I finally booted Big Sur successfully.
However, system always crashed shortly after I logged in, everything freezed and I had to reboot manually.
And I got this panic log, it showed

"Kernel Extensions in backtrace:
         com.apple.iokit.IOReportFamily(47.0)[CA61C5AE-5FC0-3924-86D7-694C4FEE5F02]@0xffffff800ee15000->0xffffff800ee17fff
         com.apple.driver.pmtelemetry(1.0)[4AF12B76-90A9-378B-9E7A-274E8DBD9240]@0xffffff7fac9e6000->0xffffff7fac9edfff".

I have no idea what it is about...... Hope someone could help me with this weird kernel panic......

[panic.log](https://github.com/CloverHackyColor/CloverBootloader/files/5361446/panic.log)

zty199, 2020-10-12:
Another new problem is that although Clover now can boot Big Sur installer successfully, after the installation finished and choose "Boot macOS from XXXXXX" for the first time, it will stuck at looping something like xpc.

I heard about this before when Big Sur Beta1 was released, and using pre-release Lilu and FakeSMC can avoid this. But now no matter I use FakeSMC or VirtualSMC it is all the same. Still trying to make it work for installing or OTA upgrading......

closed 2020-10-15 by zty199

lev666, 2020-10-27:
Hello!  I would like to know.  I have the same processor and video card.  Will your config work for my laptop?