CloverHackyColor/CloverBootloader #257: Clover hang, no idea why on 5123 big sur or catalina

bfitzgit23, 2020-10-09:
hey all,

i've spent ALL night trying to get clover and bigsur beta 9 OR catalina to boot on my Dell Inspiron 7567 laptop. I've tried the sample config.plist and one for my hardware, no change. I did import the quirks from the sample config, no fix. I am just stumped as to why it just randomly hangs via -v, i cannot make sense of this error, insanelymac forum isn't being helpful AT ALL (no surprise). So don't go telling me to go there i already tried. I have done ALL i can come up with on my own. I ran the debug option, same result. 
[image]

bfitzgit23, 2020-10-09:
[debug.log](https://github.com/CloverHackyColor/CloverBootloader/files/5354060/debug.log)

Kmbeatz, 2020-10-09:
Im also out of Luck.

I have a Kabylake Laptop with Clover 5122 no Problem. With 5123 no way to boot catalina or Big Sur.

bfitzgit23, 2020-10-09:
yeah i've been trying to fix it for weeks, no luck..i couldnt boot 5122, i only boot via OpenCore right now, but clover gives me trackpad issues, etc. never did with clover. So i'd rather use clover with bigsur

zty199, 2020-10-09:
Same situation...... I use my old config.plist and then add new Quirks part from config-sample.plist, copy those settings from my OpenCore config, but still stuck at somewhere strange, even all USB device power is cut off......

Everything works fine in 5122 which can boot Catalina installer......

SergeySlice, 2020-10-16:
The revision 5122 is too old. 
Report again with 5125

closed 2020-10-16 by SergeySlice