CloverHackyColor/CloverBootloader #258: Stuck after updating to clover 5124

syah769, 2020-10-12:
[image]

What's the problem? I updated to clover 5124 and it stuck and cannot boot to the home menu

chriseg93, 2020-10-12:
Got the same issue, it was working somewhat with the pre-release but now I'm just stuck at the apple logo.

syah769, 2020-10-12:
> Got the same issue, it was working somewhat with the pre-release but now I'm just stuck at the apple logo.

At least you're booting until apple logo. But mine do not stuck at the apple logo. Omg, what happened 😭😭

Noodle05, 2020-10-12:
Same thing here. Before it should clover menu, an error message No returned found from DataHubDxe.efi. Not sure if it's related.

mgrandinetti, 2020-10-12:
after UPDATE I lock my system to
LOG:EXITSB:START

seem a problem with bootx64.efi. If I change this all works again

thank in advance

hyperNURb, 2020-10-12:
[image]

I get the same issue

hyperNURb, 2020-10-12:
> after UPDATE I lock my system to
> LOG:EXITSB:START
> 
> seem a problem with bootx64.efi. If I change this all works again
> 
> thank in advance

How did you change the bootx64.efi?

syah769, 2020-10-12:
> after UPDATE I lock my system to
> LOG:EXITSB:START
> 
> seem a problem with bootx64.efi. If I change this all works again
> 
> thank in advance

May i know how do you change the bootx64.efi?

chriseg93, 2020-10-12:
You just download the bootx64.efi from release 5123 and replace it with the one it has currently

@hyperNURb @syah769

syah769, 2020-10-12:
> You just download the bootx64.efi from release 5123 and replace it with the one it has currently
> 
> @hyperNURb @syah769

How to replace it? Because my screen was blank

chriseg93, 2020-10-12:
My guy, I'm not an expert in this, but you should have backups just in case because you just never know what can happen..If you have no other macOS software, then you can do it through a virtual machine on Windows.

@syah769

syah769, 2020-10-12:
> My guy, I'm not an expert in this, but you should have backups just in case because you just never know what can happen..If you have no other macOS software, then you can do it through a virtual machine on Windows. That's how I would do it, not too sure if there's another way. Sorry.
> 
> @syah769

Alright, thanks for your advice

i0ntempest, 2020-10-12:
+1 stuck after updating. I get "not found returned from datahubdxe.efi", then after selecting Catalina it sticks at "[EB]#LOG:EXITBS:START]".

ifamed, 2020-10-12:
> tuck after updating. I get "not found returned from datahubdxe.efi", then after selecting Catalina it sticks at "[EB]#LOG:EXITBS:START]".

try remove datahubdxe.efi and this error will disappear

mwahh, 2020-10-12:
I am, as yet, unable to boot Catalina with either 5123 or 5124.  5123 starts to boot and then freezes early in the boot process.  5124 just shows a blank screen (after selecting the partition to boot).  In both cases I am using a config.plist that works with Clover 5118, plus the quirks section from config_sample.plist.  I've tried all the obvious things and am running out of ideas.

Hardware is a GA-Z370P D3 and a Core i5 8600 with integrated graphics.  Bios is revision F10.

mgrandinetti, 2020-10-13:
> > after UPDATE I lock my system to
> > LOG:EXITSB:START
> > seem a problem with bootx64.efi. If I change this all works again
> > thank in advance
> 
> May i know how did you change the bootx64.efi?

restored old bootx64.efui , return to pre release 5123

i0ntempest, 2020-10-13:
> > tuck after updating. I get "not found returned from datahubdxe.efi", then after selecting Catalina it sticks at "[EB]#LOG:EXITBS:START]".
> 
> try remove datahubdxe.efi and this error will disappear

Thanks, but removing this file (thus this error) still doesn't allow my OS to boot.

syah769, 2020-10-13:
> > > after UPDATE I lock my system to
> > > LOG:EXITSB:START
> > > seem a problem with bootx64.efi. If I change this all works again
> > > thank in advance
> > 
> > 
> > May i know how did you change the bootx64.efi?
> 
> restored old bootx64.efui , return to pre release 5123

To be honest, seriously I don't know how to restored the old boot efi. Sorry my my friend, hmm

i0ntempest, 2020-10-13:
@syah769 If you installed the new Clover using the pkg installer, /EFI-Backups is automatically created and all contents of your EFI folder is copied into it before the installer touches anything. You can restore it if you can gain access to your root partition. And in the future ALWAYS keep a working EFI on a USB drive or something in case something like this happens.

syah769, 2020-10-13:
> @syah769 If you installed the new Clover using the pkg installer, /EFI-Backups is automatically created and all contents of your EFI folder is copied into it before the installer touches anything. [...]

[image]

Hurm, how to restore? Any setting in this picture that I have to enter? If I click Boot MacOs, of course it will be stuck like the first picture on the top. Sorry for asking, I lost 😭😭

i0ntempest, 2020-10-13:
If you don't have another Hackintosh or Mac and are not comfortable with the scary EFI shell, there're two ways:
1. Your computer should be able to boot Windows without Clover. Once in Windows get one of the softwares that can read apfs in Windows (Paragon APFS or UFS Explorer, both have a trial period) then recover the EFI backup.
2. Take your HDD/SSD out, grab an adapter for it and a USB drive, go to somewhere that has Macs, pop them in and do the recovery.

syah769, 2020-10-13:
> If you don't have another Hackintosh or Mac and are not comfortable with the scary EFI shell, there're two ways:
>
> 1. Your computer should be able to boot Windows without Clover. Once in Windows get [...]

 

I see, alright. Thank you buddy 👍😀

zty199, 2020-10-13:
> I am, as yet, unable to boot Catalina with either 5123 or 5124. 5123 starts to boot and then freezes early in the boot process. 5124 just shows a blank screen (after selecting the partition to [...]

You can try to add boot-args "npci=0x2000" and see if it gets any help...... I successfully boot into Big Sur after adding this boot-arg, though I don't know how it works......

mwahh, 2020-10-13:
> You can try to add boot-args "npci=0x2000" and see if it gets any help...... I successfully boot into Big Sur after adding this boot-arg, though I don't know how it works......

Thanks, but no change.

SavageAUS, 2020-10-14:
If this issue is still ongoing check insanelymac clover general discussion for answers. If issue is resolved please mark it as such.

Noodle05, 2020-10-14:
Link to insanelymac discussion thread?

hyperNURb, 2020-10-14:
@Noodle05 https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?page=957

Noodle05, 2020-10-14:
@hyperNURb That is a generic discussion. Not for specific case. Easily lost in so many different topics/issues.

malhal, 2020-10-14:
I'm sad the update is broken again like it was a few months ago and I got caught out for updating without checking here first. My question why not pull the update as soon as the issue was posted?

SergeySlice, 2020-10-16:
There are several different issues mostly resolved in revision 5125.
Please don't create a thread like "I can't boot" - "I also can't".
Each issue must be specific with enough information for developers.

closed 2020-10-16 by SergeySlice

hyperNURb, 2020-10-16:
@SergeySlice sorry and thank you 🙏

SergioSuarezDev, 2020-10-25:
same here, after update blank screen on mojave

cyberfunk, 2020-10-26:
Return to release 5122 and forget you had any problems. 5123 forward is buggy and or requires fixes.

malhal, 2020-11-11:
Gave it one last chance today, upgraded from r5122 to r5126 hoping the issues were fixed but now gets stuck on small white apple logo. I'm done, switching to OpenCore.

cyberfunk, 2020-11-11:
> Gave it one last chance today, upgraded from r5122 to r5126 hoping the issues were fixed but now gets stuck on small white apple logo. I'm done, switching to OpenCore.

I highly recommend OpenCore! I've been running it for the past few weeks and its been incredible.

malhal, 2020-11-13:
Thanks I switched to OpenCore, the guides are like an infinite loop sometimes but very satisfying to get through it. I used OpenCore Configurator against their advice but I think it made it much simpler. Native startup disk selection and real disk icons in the boot chooser are a couple of things I'm enjoying.

getsueineko, 2020-11-29:
> > Gave it one last chance today, upgraded from r5122 to r5126 hoping the issues were fixed but now gets stuck on small white apple logo. I'm done, switching to OpenCore.
> 
> I highly recommend OpenCore! I've been running it for the past few weeks and its been incredible.

OpenCore is not for all. It kills my GA-Z97X-SLI and I needed recover BIOS from backup chip :)

cyberfunk, 2021-01-15:
> OpenCore is not for all. It kills my GA-Z97X-SLI and I needed recover BIOS from backup chip :)

After a few months with OpenCore I agree. It worked relatively well for a while but I noticed issues cropping up like QuickLook was breaking and other nagging problems. I've reverted back to Clover and my computer is a dream on Catalina. Rock solid. I still can't update past Clover 5122 but hoping to find a fix soon.