CloverHackyColor/CloverBootloader #260: VoodooI2c Still NOT WORK in Clover_r5124 !!

HandaTakuya, 2020-10-13:
I try update Clover 5122 to 5124 boot work BUT my touchpad (VoodooI2c) still NOR Work !! 

[image]

SergeySlice, 2020-10-16:
VoodooI2C many times reported. Test Clover 5125 or we will account the kext incompatibility with BigSur.

stevezhengshiqi, 2020-10-16:
@SergeySlice Hi Slice, VoodooI2C not work is not because of the compatibility, but related to the kext load order. We need to specify the loading order to get it to work as VoodooI2C plugins have to load earlier than VoodooI2C. Detailed explanation is at here: https://www.insanelymac.com/forum/topic/284656-clover-general-discussion/?do=findComment&comment=2739712

HandaTakuya, 2020-10-18:
> VoodooI2C many times reported. Test Clover 5125 or we will account the kext incompatibility with BigSur.

5125 not work too. 

ps. my mackintosh use 10.15.7

corint1, 2020-10-18:
the last version made voodooI2c load and work ... but the clover version is very unstable ... it boots 1 out of 10 ... and this after many changes, emotions and nerves ... reset nvram, changing the partition boot, changing boot from ssd to usb ... very unstable ....does anyone else have this problem?

JHeartyP, 2020-10-19:
Same, Voodooi2c doesn't load. Catalina 10.15.7. In kernel log I only see Voodoogpio for some reason. Kextstat shows voodooi2cservices, voodoogpio, voodoo.driver.PS2Controller, voodoo.driver.PS2Keyboard as being loaded. Voodooi2chid, voodoo input and voodooi2c does not load.

In kernel log I see this:
`2020-10-19 02:47:28.426147+0300 0xb2       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 0
2020-10-19 02:47:28.426400+0300 0xb2       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 1
2020-10-19 02:47:28.426602+0300 0xb2       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::VoodooGPIO Initializing Community 2
2020-10-19 02:47:28.426962+0300 0x74       Default     0x0                  0      0    kernel: PMRD: PowerChangeDone: ON_STATE->ON_STATE
2020-10-19 02:47:28.426976+0300 0x74       Default     0x0                  0      0    kernel: PMRD: PowerChangeDone: ON_STATE->ON_STATE
2020-10-19 02:47:28.426983+0300 0xcf       Default     0x0                  0      0    kernel: (kernel) VoodooGPIOCannonLakeLP::GPIO Controller is already awake! Not reinitializing.
`

corint1, 2020-10-19:
compile last version .... now working

bangbaew, 2020-10-20:
I also have the problem, using Clover 5125 and macOS 10.15.7 but VoodooI2C won't load, it doesn't appear in Hackintool and no trackpad found in settings, making my touchscreen unusable, any fix?

SergeySlice, 2020-10-20:
Resolved

closed 2020-10-20 by SergeySlice

wsteelenyc, 2021-01-31:
I had the same problem....Looking at my log using the command...
log show --predicate "processID == 0" --debug
I noticed this message...
Refusing new kext com.alexandred.VoodooI2C, v2.6.3: already have prelinked v2.2.
This indicated that my updated VoodooI2C.kext which is needed for the updated clover, wasn't getting installed to the /Library/Extensions/ folder.
After deleting the kext file, rebuilding my cache and rebooting I was back in business! Hopefully this information will help someone else in the future.