CloverHackyColor/CloverBootloader #261: STUCK ON EB#LOG:EXITBS

androidxda23, 2020-10-13:


androidxda23, 2020-10-13:
[image]
[image]

androidxda23, 2020-10-13:
Successfully Booted to Clover menu and Just stucks on EB EXITBS THING The inly OS IT BOOTS IS ONLY WINDOWS 10

androidxda23, 2020-10-13:
[image]

mgrandinetti, 2020-10-13:
here a Fix reading some clover Forum
in "kernel and kext patch" patch that are similare to opencore for bigger

Patch per il bootloader
Clover: KernelPM e/o KernelXCPM
OpenCore: AppleCpuPmCfgLock e AppleXcpmCfgLock

sn0wfL4ke98, 2020-10-14:
if EB#LOG:EXITBS, that means you're using the Quirks.
Try this
1. Enable SetupVirtualMap
2.
- EnableWriteUnprotector FALSE
- RebuildAppleMemoryMap TRUE
- SyncRuntimePermissions TRUE
if not, unprotector TRUE, both next are FALSE
3. Like mgrandinetti said, tick both KernelPM and KernelXCPM if your bios can't disable CFG-Lock
4. Also, tick ProvideConsoleGop

SavageAUS, 2020-10-14:
Please update this issue if it’s solved or not. If solves mark it so.

androidxda23, 2020-10-14:
still not working

sn0wfL4ke98, 2020-10-15:
what is your laptop spec?

androidxda23, 2020-10-15:
intel 8265u 
UHD 620 
SYNAPTIC TRACKPD

androidxda23, 2020-10-15:
HOW TO FIX THAT?

sn0wfL4ke98, 2020-10-15:
calm and turn off your caps lock...
- that cpu is gen 8th whiskey lake. Follow the quirks for coffee/whiskey here: https://dortania.github.io/OpenCore-Install-Guide/config-laptop.plist/coffee-lake.html#booter
- about that issue: https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/kernel-issues.html#stuck-on-eb-log-exitbs-start
- check your SSDTs and Kexts carefully. Also about that iGPU, you remember to fake id with device-id = 9B3E0000
my opinion: Always enable KernelPm and KernelXcpm. Then follow the quirks guide

androidxda23, 2020-10-15:
I did enable the kernelPM and KernelXCPM and it works but I got this another error that stucks on APFS module I had only Lilu Whatever Green and my ethernet kext to be exact 
[image]

androidxda23, 2020-10-15:
[image]

i0ntempest, 2020-10-15:
Thanks @sn0wfL4ke98 and @mgrandinetti, I can now successfully boot Catalina with 5124. The key seems to be KernelPM and KernelXCPM, at least on my system.

SergeySlice, 2020-10-16:
You have to set quirks correctly. Clover has no issue.

closed 2020-10-16 by SergeySlice