CloverHackyColor/CloverBootloader #262: Can't update Catalina to BigSur with 5124

abace18, 2020-10-13:
[image]

sn0wfL4ke98, 2020-10-13:
just about VoodooInput.kext. Remove it from kexts/Other folder, also from PlugIns folder in VoodooPS2Controller.kext

abace18, 2020-10-13:
> just about VoodooInput.kext. Remove it from kexts/Other folder, also from PlugIns folder in VoodooPS2Controller.kext

[image]

zty199, 2020-10-13:
Try to add boot-args "npci=0x2000", this may get you some help......

sn0wfL4ke98, 2020-10-14:
what is your laptop spec?

abace18, 2020-10-14:
> what is your laptop spec?

[image]

sn0wfL4ke98, 2020-10-15:
- use some common kexts like VirtualSMC, Lilu, WEG, AppleALC, VoodooPS2Controller, VoodooI2C, VoodooI2CHID, NoTouchID, SMCProcessor, SMCBatteryManager. Currently forget about AsusSMC. Also remember to remove the VoodooInput.kext like i said
- about ACPI, because that CPU is gen 8th Coffeelake, so you should need EC-USBX, PLUG, PNLF (coffeelake version) and XOSI. You can grab the prebuilt of them from Dortania OpenCore. Then try to make the config.plist again and carefully.
- about ACPI patches, only rename _OSI to XOSI is fine. Because make sure that can boot with USB
- that CPU is gen 8 but Whiskey Lake, mean the iGPU is UHD620 iirc. You have to fake the id with device-id = 9B3E0000

abace18, 2020-10-15:
> * use some common kexts like VirtualSMC, Lilu, WEG, AppleALC, VoodooPS2Controller, VoodooI2C, VoodooI2CHID, NoTouchID, SMCProcessor, SMCBatteryManager. Currently forget about AsusSMC. Also remember [...]

Tried everything you said, but directly with OpenCore, and I also got a kernel panic.

SergeySlice, 2020-10-16:
The hang at system start process means an error or incompatibility of one you kext with BigSur. 
Clover has nothing to do with this.

closed 2020-10-16 by SergeySlice