CloverHackyColor/CloverBootloader #263: Error: Not Found returned From DataHubDxe.efi

deverassilva, 2020-10-13:
after upgrading to r5124 the error (Error: Not Found returned From DataHubDxe.efi) appears at the start of the boot and then crashes

sn0wfL4ke98, 2020-10-14:
why DataHubDxe o.o? Do you use the Quirks?

DigiMaxIO, 2020-10-14:
Yeah, this happened to me as well. 

**What's happening (for me):** The error _appears during the BIOS splash screen_ before I even boot from a hard drive. I was still able to boot into the clover GUI but as soon as I tried to launch MacOS _it would just freeze_ and I had to shut it down with the power button.

**What's I learned:**At first, I researched it and discovered the Quirks section of clover had some updates, so I tested out checking all the additional recommended checkboxes from the sample plist file. But the crash still occurred so I decided to hold off on this update. 

**Result:** I had to boot into Windows and replace the BOOTX64.efi back to my r5122 version for MacOS to boot again..._oddly enough, the DataHubDxe.efi error still appears_ but everything boots just fine.

SavageAUS, 2020-10-14:
If your using 5124 or later just remove DataHubDxe.efi from drivers folder and you will be fine. 
Make a backup first.

deverassilva, 2020-10-14:
I removed the DataHubDex.efi file and the message (Error: Not Found returned From DataHubDxe.efi) stopped, but the system hangs at boot.

deverassilva, 2020-10-14:
in version r5122 the file DataHubDex.efi is present and the system works normally.

deverassilva, 2020-10-14:
This release integrated OpenCore and delegates kernel and kext injection and patching to OC which allows to boot BigSur.

PS: the configuration (System Parameters/inject Kexts) Is as "detect"

Ad3s12, 2020-10-15:
Hi there, i have exactly the same problem.

MikeManzo, 2020-10-15:
Tried to revert back to the r5122 version of BOOTX64.efi like @DigiMaxIO did.  No joy; same result.  Hangs as it tries to boot.

deverassilva, 2020-10-15:
> Tentei reverter para a versão r5122 do BOOTX64.efi como fez @DigiMaxIO . Sem alegria; mesmo resultado. Trava enquanto tenta inicializar.

I copied the backup folders (Boot and CLOVER) and returned to work

SergeySlice, 2020-10-16:
Will be resolved in 5125.

closed 2020-10-16 by SergeySlice