CloverHackyColor/CloverBootloader #264: macOS Big Sur install disk not showing up in Clover 5124

quinton-ashley, 2020-10-14:
Also I made changes to my config file on another computer and don't see those changes in Clover. Also I get the Datahubfxe.efi error too.

SavageAUS, 2020-10-14:
Do you have apfs driver?

quinton-ashley, 2020-10-14:
I have apfsdriverloader.efi  The install disk is not apfs though. I found out how to see the install disk. Must go to show hidden disks, boot the first or second one, and then Install macos big sur will show up.

quinton-ashley, 2020-10-14:
I created an OpenCore OC folder with OC Gen-X and now the install disk shows up normally but I still can't get it to boot with clover, it just shows _ at the top left of the screen.

quinton-ashley, 2020-10-14:
I didn't setup OpenCore right, missed a few steps but it's working now! I'm installing macOS Big Sur. I like it better than Clover because it's just a simple dark screen (not white) and doesn't load a bunch of windows efi boot options for all my hard drives like Clover does. OpenCore just provides a simple list with only things that can actually boot, just has one option for Windows, big improvement. Also it literally booted on the first try after I fixed the OpenCore folder.

closed 2020-10-14 by quinton-ashley