CloverHackyColor/CloverBootloader #265: Kernel panic "can't perform kext scan: no kext summary"

tahsinahmed2, 2020-10-14:
Since the pre Release v5.1 r5123 and also the Release 5124 Clover doesn't boot anymore on my PC.

It always end in a kernel panic where it says
"can't perform kext scan: no kext summary"

But on OpenCore with the same quirks config what I use on Clover it boot.
Because of that I wouldn't say it has something to do with my configuration of the quirks settings.

I've also attached a log and screenshots, maybe it helps.

https://mega.nz/file/GJYEjLoS#h6a4ubT5oDFKnQEP2311K17EInKjLWVAFFSv49oOma4

[image]
[image]
[image]
[image]
[image]
[image]

closed 2020-10-14 by tahsinahmed2

Hoangpham89, 2020-11-21:
How did you fix it ?

meimeiwe, 2020-11-24:
I've got the same problm, after I injected the another kext(for wifi driver) and snapshot by propertree