CloverHackyColor/CloverBootloader #267: Can't configure quirks on 5124

bloodstrike, 2020-10-16:
First off - it is hugely irresponsible to issue an update through the clover app that completely changes how the system works and can break systems without any warning. Something changing this major should not be a simple 'click the update button.' If you are going to change something this foundational, maybe it should come with some sort of ... you know, warning.

Now that I have a system that won't boot, I find out that 5124 requires settings in this 'quirks' section of the config. Here's what I've tried on my Coffee Lake (z390) system:

- make sure Kernel XCPM and PM are enabled
- Followed the Desktop Coffee Lake Open Core documentation and enabled:
  - AvoidRuntimeDefrag
  - DevirtualiseMmio
  - EnableSafeModeSlide
  - ProtectUefiServices
  - ProvideConsoleGopEnable
  - ProvideCustomeSlide
  - RebuildAppleMemoryMap
  - SetupVirtualMap
  - SyncRuntimePermissions

This is the best information that I can find as what I'm 'supposed' to do. However, all this gets me to is a kernel panic. I have no way currently to pull the hard drive out of this machine (NVMe drive, don't have an adapter), and this machine only has macOS on it. Is there something else I can do to attempt to get back to having a working computer?

SergeySlice, 2020-10-16:
Being in Clover GUI press letter "O" (ou) to get into Options.
Here you can find a way to switch on/off some quirks.

closed 2020-10-16 by SergeySlice

bloodstrike, 2020-10-16:
thanks for not having any reading comprehension at all. Very helpful. I can get to options to set the options I already said I was setting? Really?

MysticalOS, 2020-10-16:
Some 300 series boards don't always work with default quirks. In the same guide, look up that kernel panic in trouble shooting and you'll find alternate quirks to avoid it.
For example I was getting this issue: https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/kernel-issues.html#kernel-panic-on-invalid-frame-pointer

That, added to what slice said, play with quirks until you boot. But as I say don't do so blindly look at the different panics in trouble shooting and if you see panic you are getting it likely has alternate quirks already for you to use.