CloverHackyColor/CloverBootloader #268: WHAT MEMORY FIX SHOULD I CHOOSE?

androidxda23, 2020-10-16:
Will I use the AptioMemoryFix or the Openruntime in clover ver 5125??

mirsalimov, 2020-10-16:
I have the same question, please?

MysticalOS, 2020-10-16:
openruntime/quirks will be future going forward (with also setting correct quirks applicable to opencore guide, so for example coffee lake builds would use these quirks https://dortania.github.io/OpenCore-Install-Guide/config.plist/coffee-lake.html#booter (I realize that's an opencore guide but effectively clover is now a 3rd party front end to opencore now but effectivevly runs off same core now)

aptiomemoryfix isn't maintained anymore and not likely to work forever, especially for newer versions of macOS or newer hardware builds.

SergeySlice, 2020-10-20:
Only OpenRuntime will work for BigSur.

closed 2020-10-20 by SergeySlice