CloverHackyColor/CloverBootloader #269: 5125 boots to black screen

csoup50, 2020-10-16:
Boots correctly through clover, it correctly selects the Catalina partition and then it stops on a black screen. 
I tried 5124 and it behaved the same (halting on black screen). Im currently running 5122 and it works fine.

Z390, Radeon VII

SavageAUS, 2020-10-16:
I would suggest checking your config / kexts. Update kexts, check quirks.

mwhrtin, 2020-10-16:
Same behavior for me. 5120 booted fine, updated the kexts that had newer versions, was still able to boot after the updated kexts so I continued and updated clover to 5125. Then a black screen when trying to boot Catalina, verbose mode also just goes to a black screen. 

Have reverted back to 5120 that boots. My config.plist have been untouched for two years so unless there are any breaking changes there it "should be fine". H97, no external graphic card.

csoup50, 2020-10-16:
Thanks SavageAUS. All kexts are current. Its been a solid hack for quite a while. Something must have changed in these releases. I never tried 5123 so it could have begun then for all I know.

DecadentIpsum, 2020-10-16:
The same is happening to me as well. I am not able to boot after updating to 5125, I am getting a black screen. Any workaround will be highly appreciated :)

hyperNURb, 2020-10-16:
Try adding `-v` to the boot arguments. You should be able to see the error there.

If I understand correctly, everything after 5122 is a merge of Clover and OC, which means that you'll have to update your config. You have [sample-config](https://github.com/CloverHackyColor/CloverBootloader/blob/master/CloverPackage/CloverV2/EFI/CLOVER/config-sample.plist) and based on your motherboard/processor you need to turn on and off kexts and quirks i.e. [SkyLake](https://dortania.github.io/OpenCore-Install-Guide/config.plist/skylake.html#starting-point)

It's worth checking the [troubleshooting](https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/kernel-issues.html#stuck-on-eb-log-exitbs-start)

hapa17, 2020-10-17:
i'm having the same problem. even booting from a usb drive with unibeast with 5119 ends up in a black screen. verbose mode goes through a bunch of text then goes blank too, so i don't know exactly where the error occurs.

liyafe1997, 2020-10-18:
Me too, back screen. Even enable verbose mode also have nothing display, seems it absolutely not boot into the kernel.
And the CAPS LOCK key still response, and I can press Ctrl+Alt+Del to restart.
I had updated all the driver to the latest (from the 5125 zip)
I think it's not the kext's problem because it's not in that step. (Even not boot into the kernel yet)

hapa17, 2020-10-18:
so i got my system stable again. i think there were two problems. 

it would actually hang when booting with 5125 in verbose mode. that’s problem #1. 

problem #2 was that a unibeast boots (5119), wouldn’t seem to hang, but the screen would go black, even in verbose mode, as described previously. my hunch was that it was a graphics problem. so, i enabled integrated graphics in bios, disconnected my graphics card, booted with unibeast, and was able to reach the desktop. so i went in and rolled back clover from 5125 to 5116 (the version i had upgraded to before the problems started) to prevent the hang. 

upon reboot and connecting the graphics card again, it didn’t seem to hang, but would still go black. so i followed my hunch and having read somewhere that whatevergreen affects graphics,  i tried rolling back the whatevergreen kext, which i had updated at the same time as clover before having problems. i rolled it back from 1.4.3 to 1.4.0 i think.

after ALL that, everything works normally again. i don’t know if either of these things are known issues. my build of also an older build; not sure if that’s relevant. hope this helps someone else.

MysticalOS, 2020-10-18:
Don't all the newer AMD gpus need agdpmod=pikera to boot without black screen?

jkcoxson, 2020-10-18:
Could this be due to not having quirks set up in the plist? I also have a hang, but I didn't adjust my config at all.

jkcoxson, 2020-10-18:
Adding the default quirks has made the screen not hang at the black screen. It boots the kernel, just need to fix the memory panic now, lol. Looks like quirks are required.

liyafe1997, 2020-10-19:
> 
> 
> Adding the default quirks has made the screen not hang at the black screen. It boots the kernel, just need to fix the memory panic now, lol. Looks like quirks are required.[image]

> 
> 
> Could this be due to not having quirks set up in the plist? I also have a hang, but I didn't adjust my config at all.

I added quirks to config, I still hang at blackscreen but sometimes it can goes into the kernel, but it's very strange, everytime hang in different place (not a panic, seems it's booting normally and suddently stucked, without any error information output). Like this:
[image]

note:
The CAPS LOCK key still reseponse while blackscreen hang, and I can use CTRL+ALT+DEL to restart.
However, if it hangs after goes into kernel all key have no response (seems the system died), I only can long press the power button to turn it off.

liyafe1997, 2020-10-19:
> 
> 
> Could this be due to not having quirks set up in the plist? I also have a hang, but I didn't adjust my config at all.

I added quirks to config, I still hang at blackscreen but sometimes it can goes into the kernel, but it's very strange, everytime hang in different place (not a panic, seems it's booting normally and suddently stucked, without any error information output). Like this:
[image]

note:
The CAPS LOCK key still reseponse while blackscreen hang, and I can use CTRL+ALT+DEL to restart.
However, if it hangs after goes into kernel all key have no response (seems the system died), I only can long press the power button to turn it off.

MysticalOS, 2020-10-19:
@jkcoxson   Invlid frame pointer usually means board is on an older EDK and because of it you need to use older methods. My 370 board is like this. it's a "300 series" board but it's actually, according to firmware code in use, actually a 200 board code and usually needs older methods, so even though coffee lake in my case needs RebuildAppleMemoryMap and SyncRuntimePermissions on and EnableWriteUnprotector off. Mine cofee lake build needs literal polar opposite.

https://dortania.github.io/OpenCore-Install-Guide/troubleshooting/extended/kernel-issues.html#kernel-panic-on-invalid-frame-pointer

octoforce, 2020-10-19:
try following quirks setting!

AvoidRuntimeDefrag: YES

DevirtualiseMmio: YES

DisableSingleUser: NO

DisableVariableWrite: NO

DiscardHibernateMap: NO

EnableSafeModeSlide: YES

EnableWriteUnprotector: YES

ForceExitBootServices：NO

ProtectMemoryRegion：NO

ProtectSecureBoot：NO

ProtectUefiServices：NO

ProvideCustomSlide：YES

RebuildAppleMemoryMap：YES

SetupVirtualMap： NO

SignalAppleOS： NO

SyncRuntimePermissions：YES

SergeySlice, 2020-10-20:
Dear sirs,
Issue tracker is not a place to ask how should I set quirks.

closed 2020-10-20 by SergeySlice