CloverHackyColor/CloverBootloader #271: Can’t boot Big Sur in clover after updating to beta 10

jzrodriguez98, 2020-10-16:
[image]
After﻿ updating to beta 10, I checked if clover boots Big Sur and it did not work. I tried with clover v5123 and 5125 and none of them worked. I updated all kexts. 
[Archive.zip](https://github.com/CloverHackyColor/CloverBootloader/files/5394680/Archive.zip)
It was working with beta 9.﻿ I’m getting a memory panic when booting. Could you check if there’s something wrong? My build is a GA-Z97-HD3, RX-580 8GB, 16GB RAM

MysticalOS, 2020-10-16:
Beta 10 requires new OC changes, even official OC 0.6.2 doesn't work, requires 0.6.3 alpha with change they made in last 24 hours. Clover will need to pickup those mainline changes as well.

jzrodriguez98, 2020-10-17:
@MysticalOS thanks a lot for the feedback. I’m using OC 0.6.3 to boot beta 10, so I understand what you are saying.

jzrodriguez98, 2020-10-17:
@MysticalOS, I was able to boot from the preboot but the wifi is not recognized (neither wifi device properties included in config file are injected nor the DSDT wifi serialization is recognized). I was expecting that to happen based on others’ comments about booting BS from Preboot.

nooretk, 2020-10-17:
So now how to fix this?
I'm unable to boot to the installer

SavageAUS, 2020-10-17:
Have patience.

SergeySlice, 2020-12-21:
Clover 5127 booted the installer without problem.

closed 2020-12-21 by SergeySlice